arendst/Tasmota #1078: Buttontopic remote switch

dawsmac, 2017-10-27:
Hey all need a little pointer, with my Sonoff Basic. 

I have it setup with a real switch connected to GPIO14 and have linked this to Button2 in the menu.
I have then set the Buttontopic to "remoterealswitch".

If I have understood this right when triggering the switch on GPIO14 should I not see this in the console and then the MQTT broker and HA receive this information 
 
any help would be great...

dawsmac, 2017-10-27:
Just a little help would be great... I have read all the wiki about 5 times but not really understanding the Buttontopic ;)

davidelang, 2017-10-28:
buttontopic makes it so that none of the buttons will control the local relays, 
instead they will just send MQTT messages.

if you want the built-in button to work as normal, define the thing you hve 
on GPIO14 as a switch, not a button, and set switchtopic, but undo button topic.
k

dawsmac, 2017-10-28:
Hi Dave, 

cheers for your input, I do want the external switch to just send a MQTT message to my broker that will get picked up by HA. I might have it wrong. I have setup GPIO14 as "Button2" and set up a buttontopic.. but when turn the switch off and on I don't see any message in console

davidelang, 2017-10-28:
set debugging to 'more detail' and you will see something happen when you turn 
the switch on and off.

If you want to do something when it's turned on and something different when 
it's turned off, you need to set it as a switch and set the appropriate 
switchmode.

a button is very limited, it's designed to be a momentary switch (like what is 
built-in on the devices)

closed 2018-05-07 by stale[bot]