arendst/Tasmota #1080: How to see all possible networks with RSSI information?

stefanbode, 2017-10-27:
I have some Fritz Repeater in my home repeating the SAME WLAN SSID. would be great to have a "scan" function with SSID and MAC-Adress and RSSI to see which repeaters are nearby and how clear is the signal. I assume that on connection the internal procedures will always use the strongest network on reboot/reconnect when finding more than one network with the same name. 

Thanks for sharing more information.

jetema, 2017-10-27:
There's a scan function via web, hope it helps 
[image]

stefanbode, 2017-10-28:
No, this neither show the MAC-Address, nor does it show SSID that where broadcasted by more than one repeater. In this case the Ssid is the same and the MaC is different.

stefanbode, 2017-11-02:
There is a hard-coded option to enable/disable this in webserver.ino

`boolean remove_duplicate_access_points = false;`

Good to know.

closed 2018-05-07 by stale[bot]