arendst/Tasmota #1081: Update from 4.0.8 to latest version failed

mittelfeldmotor, 2017-10-27:
I have several sonoffs with tasmota 4.0.8
I tried to update but it failed. I tried several sonoffs Mord than one time. Is there a solution without flashing?

jetema, 2017-10-27:
If you don't want compiling, you could try flashing only the binary file ( sonoff.bin ) available on 
https://github.com/arendst/Sonoff-Tasmota/releases

mittelfeldmotor, 2017-10-27:
I solved it. My way was first backup config, then ota update 5.0. then update to newest version.Now all sonoff habe the 5.8.0

closed 2017-10-27 by mittelfeldmotor

davidelang, 2017-10-28:
I seemto remember that there was a break early in the 5.x series where you have 
to flash from earlier versions to that 5.x version, and then from there to the 
current version. I don't remember where that break was (and I need to update a 
few @arendst what was that point?)

arendst, 2017-10-28:
As far as I remember the info on the upgrade page https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade is still valid.