arendst/Tasmota #1082: Sonoff Basic Sensor issue

edit21, 2017-10-27:
Hi all, Have two basics, one on 5.8.0m and the other on 5.5.2b
Using a dht22 on both - both have identical setup though the on on 5.8.0m shows no sensor output, not even the sensor line in console, hoping this was a misconfiguration but all looks the same .

Working Basic:
20:34:25 MQTT: tele/sonoffsingle1/STATE = {"Time":"2017-10-27T20:34:25", "Uptime":34, "Vcc":3.165, "POWER":"ON", "Wifi":{"AP":1, "SSID":"BT-OpenRelay", "RSSI":70, "APMac":"00:1D:AA:F4:85:30"}}
20:34:25 MQTT: tele/sonoffsingle1/SENSOR = {"Time":"2017-10-27T20:34:25", "DHT22-14":{"Temperature":20.1, "Humidity":54.6}, "TempUnit":"C"}

Non working Basic (5.8.0m)
00:20:24 MQT: tele/sonoffsingle2/STATE = {"Time":"1970-01-01T00:20:24", "Uptime":1, "Vcc":3.138, "POWER":"OFF", "Wifi":{"AP":2, "SSId":"Error", "RSSI":40, "APMac":"00:1A:8C:C0:E4:31"}}

notice the MQT after the time stamp, all other logs are 3 chars long - not sure if that is relevant

arendst, 2017-10-27:
Try the latest version and enable more logging with option 3.

edit21, 2017-10-28:
Thanks - will do!

edit21, 2017-10-28:
Working now as expected on new FW \o/ 
TYVM for the code Arendst!

closed 2017-10-28 by edit21