arendst/Tasmota #1083: Use the RX/TX pins as MQTT-serial passthrough?

czyz, 2017-10-27:
I'm planning to use a Sonoff Basic to power up a rotating disco ball and light when a button is pressed. I was planning to have a GPIO pin on the ESP8266 trigger an arduino to send serial commands to a Catalex serial-controlled MP3 player (model YX5300. Link to [product](http://www.dx.com/p/uart-control-serial-mp3-music-player-module-for-arduino-avr-arm-pic-blue-silver-342439), [manual](https://github.com/salvadorrueda/ArduinoSerialMP3Player/blob/master/Serial%20MP3%20Player%20v1.0%20Manual.pdf)). Then I remembered that the ESP8266 has its own RX and TX pins, and figured maybe I could skip the arduino entirely and have the ESP8266 from the Sonoff Basic control the Catalex. The Catalex module accepts commands like "7E FF 06 03 00 00 01 EF" (which plays the first track) over a 9600 8-N-1 connection.

Thinking about using the RX and TX pins on the Sonoff's ESP8266 for serial communication reminded me that I have many devices other than the Catalex that can be controlled over serial. And this led to an idea that's more general-purpose than controlling that mp3 player module. What if one could send a string, such as the aforementioned "7E FF 06 03 00 00 01 EF" to a Sonoff Basic over MQTT, and have the Sonoff output that command over its serial pins? Is that doable? Generalized serial relaying over MQTT could be very useful for controlling not only mp3 modules but many pieces of AV equipment. I'd love to be able to send serial commands from a node-red flow to be relayed to various items.

I just figured I'd float the idea here so that you more knowledgeable folk could either shoot it down as instantly impractical/impossible or possibly find it worthy of further thought. It's possible that the Sonoff devices make use of those TX and RX pins and they're thus unavailable for serial, or there wouldn't be enough memory to implement this as a feature, or the debugging output from the chip takes total control of serial communications.

ionciubotaru, 2017-10-28:
Yes, you cand do this
ESP8266 TX/RX works just like ATMEGA368. All you have to do is to subscribe via MQTT, get received messages and send it via serial.

gheesung, 2017-10-29:
I have done a DIY RF bridge which I have connected an UNO to ESP-01 using Tasmota firmware. The concept can be extended to your application. Just need to configure Tasmota as RF bridge and the RF Bridge is sending a code via the RX/TX to the connecting device when one of the RF(x) button is pressed or a message is received via the MQTT e.g devicetopic/cmnd/rfkey1. To send a specific code, you may need to modify the Tasmota firmware or use the learn mode.

The source code can be found at [github](https://github.com/gheesung/sonoffRFbridge).

czyz, 2017-10-29:
This sounds very promising, but I'm a little confused. Is the ESP-01 sending serial commands to the Uno, and then the UNO is itself directly controlling the FS1000A?

I think what I'd like to do is add to the Tasmota firmware and configurations menus the ability for a Sonoff Basic to be used as a MQTT->serial communications relay. So I've made a branch called "Sonoff-Tasmota-serialrelay" and am now trying to figure out where to modify the code to add the serial relay function to the configuration menu and how to get ahold of MQTT payloads and serialwrite them out the TX pin. No additional hardware required. 

I may be able to get my needs met by utilizing the IRRemote feature to send a few preset commands, but that doesn't feel simple enough to be generalized for just about any serial communications purpose. Not that I'm sure I can manage to construct such a thing, but I'm giving it a try.

gheesung, 2017-10-30:
Tasmota is communicating with the external device via serial (19200 baud) when configured as Sonoff RF bridge.  In my case, Uno is controlling the 433 transmitter and receiver. 

As Tasmota is mainly served an alternative firmware for Sonoff devices, you may have to look at other firmwares to achieve your objective or start changing the code.

closed 2018-05-07 by stale[bot]