arendst/Tasmota #1085: IRsend Command????

Bluelinux, 2017-10-28:
how can I send an IR code? I have already tried many commands, but without success.

Example: irsend {"protocol": "SAMSUNG", "bits": 32, "data": 551502015}

venugopalsomu, 2018-03-30:
You need to send to topic thats end with /irsend
the data should be {"protocol": "SAMSUNG", "bits": 32, "data": 551502015}

eps0, 2018-06-06:
I am trying to send an ir code through mqtt. I can send it through the console using:  IRsend { "Protocol": "SAMSUNG", "Bits": 32, "Data":0xE0E0E01F} and I get back: 15:04:18 MQT: stat/sonoff14/RESULT = {"IRSend":"Done"}. I can also send the code with this link: http://192.168.1.20/cm?cmnd=IRsend { "Protocol": "SAMSUNG", "Bits": 32, "Data":0xE0E0E01F} and everything works fine. I just can't figure out how to send it through mqtt and what topic to send it to.

ascillato, 2018-06-13:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#irremote

closed 2018-06-13 by ascillato2