arendst/Tasmota #1087: v5.8.0 Proplem with Compilation

Bluelinux, 2017-10-28:
Arduino: 1.8.2 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

Build-Optionen wurden verändert, alles wird neu kompiliert
xdrv_ir_send:50: error: 'IRsend' does not name a type

 IRsend *irsend = NULL;

 ^

D:\Downloads\Sonoff-Tasmota-development-5.8.0\Sonoff-Tasmota-development\sonoff\xdrv_ir_send.ino: In function 'void IrSendInit()':

xdrv_ir_send:54: error: 'irsend' was not declared in this scope

   irsend = new IRsend(pin[GPIO_IRSEND]); // an IR led is at GPIO_IRSEND

   ^

xdrv_ir_send:54: error: expected type-specifier before 'IRsend'

   irsend = new IRsend(pin[GPIO_IRSEND]); // an IR led is at GPIO_IRSEND

                ^

xdrv_ir_send:54: error: expected ';' before 'IRsend'

D:\Downloads\Sonoff-Tasmota-development-5.8.0\Sonoff-Tasmota-development\sonoff\xdrv_ir_send.ino: At global scope:

xdrv_ir_send:73: error: 'IRrecv' does not name a type

 IRrecv *irrecv = NULL;

 ^

D:\Downloads\Sonoff-Tasmota-development-5.8.0\Sonoff-Tasmota-development\sonoff\xdrv_ir_send.ino: In function 'void IrReceiveInit()':

xdrv_ir_send:78: error: 'irrecv' was not declared in this scope

   irrecv = new IRrecv(pin[GPIO_IRRECV]); // an IR led is at GPIO_IRRECV

   ^

xdrv_ir_send:78: error: expected type-specifier before 'IRrecv'

   irrecv = new IRrecv(pin[GPIO_IRRECV]); // an IR led is at GPIO_IRRECV

                ^

xdrv_ir_send:78: error: expected ';' before 'IRrecv'

D:\Downloads\Sonoff-Tasmota-development-5.8.0\Sonoff-Tasmota-development\sonoff\xdrv_ir_send.ino: In function 'void IrReceiveCheck()':

xdrv_ir_send:89: error: 'decode_results' was not declared in this scope

   decode_results results;

   ^

xdrv_ir_send:89: error: expected ';' before 'results'

   decode_results results;

                  ^

xdrv_ir_send:91: error: 'irrecv' was not declared in this scope

   if (irrecv->decode(&results)) {

       ^

xdrv_ir_send:91: error: 'results' was not declared in this scope

   if (irrecv->decode(&results)) {

                       ^

D:\Downloads\Sonoff-Tasmota-development-5.8.0\Sonoff-Tasmota-development\sonoff\xdrv_ir_send.ino: In function 'boolean IrSendCommand(char*, uint16_t, char*, uint16_t, int16_t)':

xdrv_ir_send:305: error: 'irsend' was not declared in this scope

             irsend->sendNEC(data, bits);

             ^

xdrv_ir_send:307: error: 'irsend' was not declared in this scope

             irsend->sendSony(data, bits);

             ^

xdrv_ir_send:309: error: 'irsend' was not declared in this scope

             irsend->sendRC5(data, bits);

             ^

xdrv_ir_send:311: error: 'irsend' was not declared in this scope

             irsend->sendRC6(data, bits);

             ^

xdrv_ir_send:313: error: 'irsend' was not declared in this scope

             irsend->sendDISH(data, bits);

             ^

xdrv_ir_send:315: error: 'irsend' was not declared in this scope

             irsend->sendJVC(data, bits, 1);

             ^

xdrv_ir_send:317: error: 'irsend' was not declared in this scope

             irsend->sendSAMSUNG(data, bits);

             ^

xdrv_ir_send:319: error: 'irsend' was not declared in this scope

             irsend->sendPanasonic(bits, data);

             ^

exit status 1
'IRsend' does not name a type

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

wunderbaum, 2017-10-29:
Seems to be wrong  livrary version.

Bluelinux, 2017-10-29:
from IRRemoteESP8266?

u2mike, 2017-11-20:
same issue ... any hints? (detailled)

plReynaerde, 2017-11-21:
See: https://github.com/markszabo/IRremoteESP8266/releases

Add two lines underneath #include <IRremoteESP8266.h>
saying:
#include <IRsend.h>  // If you want to send IR commands.
#include <IRrecv.h>  // If you want to receive IR commands.

See at the bottom of the mentioned page.

u2mike, 2017-11-21:
thx for your reply ... i started from the scratch with the following link:
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE
NOW i can compile but problem is to upload to my sonoff pow device:
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
so now i am trying to iron out details about this issue :-(

i checked cable and so on but until now uploading doesn´t work!

closed 2018-05-07 by stale[bot]