arendst/Tasmota #1088: Sonoff Touch: Slow reaction

TylerDurden23, 2017-10-28:
The reaction to touching the Sonoff Touch is relatively slow - almost 1 second from touching to the relay being switched.

jeylites, 2017-10-28:
@TylerDurden23 

That’s because tasmota lets you set different things if you press the button
multiple times.  You could disable this feature by issuing the command "setoption13 1" 
It was covered before, see below 
[issue 891](https://github.com/arendst/Sonoff-Tasmota/issues/891)

TylerDurden23, 2017-10-28:
Thanks for the fast reply, confirmed that this solves the issue.

Should still be the default for the touch I think, since the 1 second delay is quite long - and that after you remove the hand from the switch.

closed 2017-10-28 by TylerDurden23

davidelang, 2017-10-28:
If it was the default, you would not be able to do a 4 second press to reset 
your config. You also would not be able to use any of the multipress options.

but you can make it the default for you by compiling your own image with the 
default set the way you want it.