arendst/Tasmota #1093: Commands Ignored

lespatots, 2017-10-29:
I'm trying tochange the Wifi and also the MQTT broker using the Web interface. I can save but when the module reboots the change hasn't stuck. I also tried to change the MQTT server via the console and again it seems to be accepted but doesn't stick?

Am I missing somethind, more than on module does this. I am on 5.5.5.

davidelang, 2017-10-29:
did you flash the minimal image? that doesn't allow you to save any settings, 
it's only there to let you flash an image larger than 1/2 your flash size.

lespatots, 2017-10-29:
No I don’t think so, in any event I downloaded the firmware update using OTA? Would that resolve it if I had?

davidelang, 2017-10-29:
it all depends on which firmware image you flashed, not how you flash it.

lespatots, 2017-10-29:
David, Thanks, doesn't 5.5.0 define the version? I'm not clear what you get OTA from http://domus1:80/api/arduino/. It wasn't a later release, which surprised me. Is the answer just to flash the latest firmware or can I do an OTA with the right URL?

lespatots, 2017-10-29:
> no, there are multiple .bin images that are avaialble pre-compiled, you can have
> 5.8 minimal as well as full images
>
> http://domus1:80/api/arduino/ isn't a fully qualified domain name, that would
> have to point at something on your network, so it would be what you compiled.
> Did you turn on BE_MINIMAL in your user_config.h?

#define BE_MINIMAL, so I guess that sets it on. If I point OTA at a version without this set can I do an OTA upgrade? It isn’t clear in the code comment what this does and if it is elsewhere in the Wiki I missed it. I guess it is back to the flash board and redo this. At lease I can set the params I wanted to change into the starting image!

Thanks for the help

davidelang, 2017-10-29:
BE_MINIMAL disables most features, including the ability to save configs.

The purpose of it is to create an image that has just enough functionality to be 
able to flash another image.

old image size  + new image size must be < flash size

If your image size is > 500m, you cannot flash another image that same size, so 
the be_minimal config is a quick way to create a small image that you can flash 
and then flash a full image

so comment out BE_MINIMAL and compile a new image, then flash it (and while you 
are at it, I would upgrade from the 5.5 that you are compiling to the current 
version)

lespatots, 2017-10-29:
I’ll do that and now you explain the purpose of BE_MINIMAL it is obvious, just that this isn’t what the comment in the config’h file says. I seem to remember now that finding the compiled image is a trifle complicated under OSX but if I’ve done it before it can’t be impossible.

Many thanks again for you rapid help

closed 2018-05-07 by stale[bot]