arendst/Tasmota #1096: Multisensor (temp/hum) does not send idx1, only temp

Tresett, 2017-10-30:
Hi.

TH16+AM2301 (sensor for Temperature and Humidity).
Main page show Temperature value and Humidity value.

I'm configure Domoticz page:
idx from Domoticz to idx1 (Temp only). Domoticz does not get this value.
next one idx from Domotics to idx2 (Temp and Humidity). Domoticz get this value.

Why Tasmota does not sent IDX1 value?

arendst, 2017-10-30:
The AM2301 is a temperature AND humidity sensor. So you select a Domoticz index for a Temperature AND Humidity sensor as that is the only selection supported by the AM2301 driver.

I thought that makes sense but who am I...

Tresett, 2017-10-30:
Domoticz scripts does not work correctly with multidata without conversion.
Blocky doesn not work (very simple language =).
I make explode and converto to int on LUA script. But I hope to drop this code after change FW from ESPEasy to Tasmota.
It turned out that Tasmota does not work that way, it's a great pity.

closed 2017-10-30 by arendst