arendst/Tasmota #1097: 4ch pro ticks relay on boot

jscurtu, 2017-10-30:
I have a problem with my 4ch pro running tasmota, I am using one channel as a front door opener.
Unfortunately, every time the device is rebooted or a different error occurs (Network etc) the relay is switched ON for just a second while reinitializing which of course will unwillingly open the front door. 

I have searched everything but could not find a solution for this.. Is there anything I can do in the firmware to prevent this behavior?

My dip switches are set as so:
S6 to 1
K5 1110
K6 0001

Thanks

ionciubotaru, 2017-10-30:
It's a hardware problem. Solutions:
1. use the NC pins of the relay
2.add a capacitor on the relay pins big enough to keep the relay's state

jscurtu, 2017-10-31:
When you say use NC pins, what do you mean? 
I have 12V DC going in the NC pin and COM for the switch ..

ionciubotaru, 2017-10-31:
a relay has 3 contacts NC - common - NO 
NC = normal close
NO = normal open

The relay contacts must be potential free in order to use it for your garage door

bru1979, 2017-11-27:
@jscurtu does it also happen when removing power to the 4CH Pro? 

Here it happens only when hitting RESTART button, not when unplugging power jack and plugging it back. I'm already avoiding soft restarts on network problems using **WifiConfig 4** [https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#wifi](url)

Could it be a firmware issue and not hardware?

closed 2018-05-07 by stale[bot]