arendst/Tasmota #1098: Template with "Maxi module"

flexiti, 2017-10-30:
Hi ,
Can we add to template full configurable module ?? eg.

  { "Maxi module",     
     GPIO_USER,        
     GPIO_USER,        
     GPIO_USER, 
     GPIO_USER,       
     GPIO_USER,  
     GPIO_USER,       
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_USER, 
     GPIO_USER, 
     GPIO_USER,       
     GPIO_USER, 
     GPIO_USER,        
     GPIO_ADC0         // ADC0 A0 Light sensor / Requires USE_ADC_VCC in user_config.h to be disabled
  },
It would be nice if the analog could also be set up  :smiley:

arendst, 2017-10-30:
How about the already present Wemos D1 Mini...

flexiti, 2017-10-30:
Now I see, its ok, only name suggests a specific module
What about GPIO_ADC0 on/off ??

arendst, 2017-10-30:
Impossible without recompile. See Arduino-esp8266 core site.

flexiti, 2017-10-30:
I have apk, if it receives "Analog" displays the control. So I have to compile for one module without the analog option. (I only use Witty modules) since no analog display is required for this particular module. I can always work around this with other methods, but I thought that if I could set it in the module configuration would be best 😃 thanks

closed 2017-10-30 by arendst

flexiti, 2017-11-02:
The solution for not compiling was pretty easy, I created another  "Maxi module",  with

 0 // ADC0 Analog input  , instead GPIO_ADC0  and analog data is not send any more 😃