arendst/Tasmota #1099: Request : Add Support to PZEM 004T

gajotnt, 2017-10-30:
Looks way safer to use in high wattage equipements than the sonoff POW.

https://github.com/olehs/PZEM004T

https://www.aliexpress.com/item/PEACEFAIR-AC-100A-Electric-power-monitoring-and-communication-module-power-meter-power-energy-Volt-Ammeter-with/32405328185.html?aff_platform=product&cpt=1509366879543&sk=F2zZRNv&aff_trace_key=ae2cbe39517f4abe965107629801917c-1509366879543-05026-F2zZRNv&terminal_id=a987af0fbdf4413dbb87d979911ed0d8

https://easydomoticz.com/forum/viewtopic.php?f=24&t=4687

gajotnt, 2017-12-12:
I know this is not your problem because this is from a PR that adds support for a energy sensor (https://github.com/arendst/Sonoff-Tasmota/pull/614), everything is working except it doesnt send updates to domoticz, the idx is properly set in the domoticz page, maybe someone can take a look in the code and give me some light :( the PR coder isnt responding to me.
Would be awesome if PZEM004T had official support

```
09:47:32 MQTT: tele/sonoff/ENERGY = {"Time":"2017-12-12T09:47:32", "Yesterday":0.000, "Today":5.265, "Period":2, "Power":225.0, "Voltage":238.8, "Current":1.110}
09:48:02 MQTT: tele/sonoff/STATE = {"Time":"2017-12-12T09:48:02", "Uptime":0, "Vcc":3.448, "Wifi":{"AP":1, "SSID":"TNT", "RSSI":40, "APMac":"9C:97:26:9F:A5:0F"}}
09:48:02 MQTT: tele/sonoff/ENERGY = {"Time":"2017-12-12T09:48:02", "Yesterday":0.000, "Today":5.267, "Period":2, "Power":228.0, "Voltage":236.9, "Current":1.120}
09:48:32 MQTT: tele/sonoff/STATE = {"Time":"2017-12-12T09:48:32", "Uptime":0, "Vcc":3.447, "Wifi":{"AP":1, "SSID":"TNT", "RSSI":38, "APMac":"9C:97:26:9F:A5:0F"}}
09:48:32 MQTT: tele/sonoff/ENERGY = {"Time":"2017-12-12T09:48:32", "Yesterday":0.000, "Today":5.269, "Period":2, "Power":225.0, "Voltage":238.8, "Current":1.110}
```

closed 2017-12-28 by gajotnt

brlac, 2018-01-12:
I saw this post and support for energy sensor (#614) in Sonoff basic. I do have pair of Sonoff basic and was wondering how PZEM 004T would get connected. Can Sonoff basic be used as a switch and read the sensor data from TTL interface? Is it safe to connect tx/rx if Sonoff is connected to the mains? Last one is how to provide 5v to energy sensor and 3.3v to sonoff. I was trying to picture how the devices would connect to each other as the sensor seems to be accurate at measuring the energy use. Thank you.

gajotnt, 2018-01-12:
Yes you can use the relay of the basic and use the pzem. 
Yes it is safe :)
The 5v when I get to a PC I will post a picture of were you can connect.

brlac, 2018-01-12:
Thanks for your response and time :) That would be great if you can share the pictures on how things are connected.

sdhawade, 2018-02-04:
@gajotnt Can you please share the connection options for SonOFF Basic  ?

afaucogney, 2018-09-11:
@gajotnt any news about the picture ?