arendst/Tasmota #1101: KRACK Wifi vulerability and 2.4.0-rc2 update

stefanbode, 2017-10-30:
Hi @arendst, the ESP8266 is/was also vulnerable to the KRACK attack and Espressif quite fast fixed the issue with the 2.4.0-rc2 release. Is there anything I need to change on my ATOM development environment to get sketched build with the new fixed code. I know I have to replace ALL IoT devices with the new code.

Thanks

arendst, 2017-10-30:
Try to  get a supported 2.4.0rc2 version with the krackfix installed. As far as I can tell this fix is not available in 2.4.0rc2 yet as it still uses an old sdk.

If you compile now with 2.4.0rc2 prepare for hardware watchdog reboots as it uses 10k more memory than 2.3.0 does.

stefanbode, 2017-10-31:
10k memory is quite a lot. Especially because i normally run it with TLS MQTT as you know. This will be tough. Maybe I have to optimize my Webserver usage to get around it. I was not aware that I installed 2.3 anywhere. I know how to do it in the boardmanager and library in Arduino ide, but how this works in Atom. To be honest: look magic for me right now. In the great YouTube video from Andreas Spieß he mentioned that the vulnerability is fixed now. Also the commends from Igor from Espresssif looks like this.

arendst, 2017-10-31:
You're right. It seems the released version 2.4.0RC2 indeed contains the Krack fix. 

A pity it still uses too much memory to stay reliable as noted her too https://github.com/esp8266/Arduino/issues/3740

stefanbode, 2017-11-01:
Ok, interesting challenge to get this working...
First I added the staging version with: 
`$ platformio platform install https://github.com/platformio/platform-espressif8266.git#feature/stage`
Then I changed in the platform.ini for my wemos D1 mini devices the platform from old:
```
[env:wemos]
;platform = espressif8266
platform = espressif8266_stage
```
Nor compilation to some strange things. Anhow at the end my build is different. Maybe correct maybe not. Here the differences:
Programm Size: 511kb -> 543kb (not a problem for me, but mentionable)
Free memory 22k -> 25k (Uups this I was not expecting. incl Webserver. excl. TLS MQTT.
The Info Page hint to a new version but not as I was expecting:
OLD: 
`Core/SDK Version | 2_3_0/1.5.3(aec24ac9)`
NEW:
`Core/SDK Version | 00000000/2.1.0(deb1901)`

I assume there are still mistakes in it. Any help really welcome, before I update all my devices.

stefanbode, 2017-11-03:
@arendst or @davidelang: can you give some insights, if this above is sufficient, or if there are additional steps required. Also, a hint regarding the RAM would be great. Is it only me that see a benefit?

closed 2018-05-07 by stale[bot]