arendst/Tasmota #1102: Problem with switch performance and button

freaktech, 2017-10-30:
Hi!

I am total new to Sonoff and flashed 1 basic device today with the FW 5.9
i am not a developer and have basic electric skills.

i notice that the performance of the webinterface is very slow :-(  ok may be the hardware 

my real problems are ......

Problem 1:
i want to connect a standard switch to GPIO14
it is working just 2-3 times in realtime, then i have a delay of about 10-15 seconds?

Problem 2:
i want to connect at a different project a button (not a switch)
the button should just toggle the relay every time from on to off
i tried the settings button   but none of the shows a section

here is a video with the switch

http://www.youtube.com/watch?v=sw4B1j8-h8c

i also used a 1K resistor between 3.3V and GPIO14, no difference

thank you 

tom

davidelang, 2017-10-30:
turn on detailed logging, I'll bet that you find that the device is attempting 
to establish a connection to the MQTT server at the time you are having the 
delay. This is a bug in the TCP library for Arduino, while it is trying to make 
a connection, nothing else can take place.

freaktech, 2017-10-31:
Hi david!

thank you for your help.

1. i don't know what is MQTT    i just want to use ECHO  
2. have no idea how to log

BUT

when i disable MQTT   the external switch is working perfect and the toggle function at the Webinterface is very fast also
and it looks like Alexa is working in HUE mode also now

button / no switch:
i still have no idea how i get a button at GPIO14 working
1 x short press ON
1 x short press off

thank you tom

[image]
[image]
[image]

freaktech, 2017-10-31:
i want also to use a momentary switch button   not a switch with permanent open or close position

1 klick = stable ON 
1 klick again = OFF

als long as you click again

how is this possible

1 resistor between 3.3V and GPIO14  brings no difference

thank you tom

davidelang, 2017-10-31:
please stick to one issue for your question rather than asking the same thing 
in two places.

closed 2018-05-07 by stale[bot]