arendst/Tasmota #1103: sonoff 4ch pro - question for config

roblad, 2017-10-31:
Hi,

I have a question regaring config and gpio, on chematic I see that there is no flash connection on
GPIO 7,8,6,11,16  for moule PSF-A85

https://www.itead.cc/wiki/images/f/fe/Sonoff_4CH_Pro_Schematic.pdf

I want to buy it and add T11 and 5 x DS18x20, and may be use I2C for BMP85  - could you explain a little bit why there are zeros in template ? 

in your setting there are 0 and describe as flash connection, there is some mistake ?

{ "Sonoff 4CH Pro",  // Sonoff 4CH Pro (ESP8285)
     GPIO_KEY1,        // GPIO00 Button 1
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_USER,        // GPIO02 Optional sensor
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_REL3,        // GPIO04 Sonoff 4CH Red Led and Relay 3 (0 = Off, 1 = On)
     GPIO_REL2,        // GPIO05 Sonoff 4CH Red Led and Relay 2 (0 = Off, 1 = On)
     0, 0, 0,          // Flash connection
     GPIO_KEY2,        // GPIO09 Button 2
     GPIO_KEY3,        // GPIO10 Button 3
     0,                // Flash connection 
     GPIO_REL1,        // GPIO12 Red Led and Relay 1 (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     GPIO_KEY4,        // GPIO14 Button 4
     GPIO_REL4,        // GPIO15 Red Led and Relay 4 (0 = Off, 1 = On)
     0, 0

ionciubotaru, 2017-10-31:
NO, this is not a mistake, the pins are for eeprom communication 
ESP8285 has eeprom in the same chip, but it is a esp8266 + eeprom, so those pins cannot be used as GPIO

daspid92, 2017-10-31:
I'm actually using GPIO16 for an External Switch I've only changed the configuration in the template replacing the 0 with a GPIO_USER, added a pull up resistor and soldered a wire on the PSF pin, it's perfectly working. I've also defined the pins previously used for Button2/3/4 as GPIO_USER and used them as Switches, and also some others PINs not attached to anything in the schematic for future sensors i would to add.

roblad, 2017-10-31:
Hi,

Are you sure ?

ESP8285 has eeprom in the same chip, but it is a esp8266 + eeprom, so those pins cannot be used as GPIO

why then they specyfied on shematic sheet even that the DS18x20 is ready, and pins not connected to any chip ?

https://www.itead.cc/wiki/images/f/fe/Sonoff_4CH_Pro_Schematic.pdf

Did you test it, that it has been connected to internal SPI ?

What exactly pins can be used as additional then ?  

GPIO_USER, // GPIO01 Serial RXD and Optional sensor
GPIO_USER, // GPIO02 Optional sensor
GPIO_USER, // GPIO03 Serial TXD and Optional sensor
GPIO_USER // GPIO16
ADC

arendst, 2017-10-31:
Dig a bit deeper regarding the PSF-A85 on the itead wiki website and you see:

[image]

The pins 6,7,8 and 11 are used as SPI connection to the internal EEPROM

daspid92, 2017-10-31:
I'm using GPIO9/10/14/16 for Switches and they are working, and I've already configured GPIO6 and 11 as GPIO_USER but this two not already tested

roblad, 2017-10-31:
Hi,
to: daspid92

Could you check, if that pins can be used if you have that module ?

Thanks,
RL

edit21, 2017-11-05:
Hi All,
I only see GPIO 1,2 & 3 in the configure module page and have been hunting for the breakout locations of this, I too would like to add a sensor :)

daspid92, 2017-11-10:
@roblad Sorry I was not at home, I confirm that GPIO16 is working, for the other two GPIOs if they are connected to the internal eeprom I don't know if it's safe to use them, I've now removed them from my config as GPIO_USER....

roblad, 2017-12-14:
Hi,

Could you somebody can confirm if GPIO7 and GPIO8 according sonoff 4ch can be used as TEMP1 and TEMP2 in sonnoff 4ch pro, what is a differences when the same PSF-A85 chip is used and for sonoff 4ch pro as @arendst said is used for internal flash ? Does sonoff 4ch have no internal flash :-) ? See on schematic for sonoff 4ch

https://www.itead.cc/wiki/images/d/d3/Sonoff_4CH.SCHMATIC.pdf

imahmud003, 2018-02-21:
Hi All,
I want to use the 4CH pro for my garage door which has separate push buttons for UP and DOWN and also pressing either button when in motion works as STOP.  
I am using Hassio with internal MQTT broker. So I will connect the three relays (inching mode) to use the three modes. 
My question is that I want to also connect a wired reed switch to monitor the state of the door. 
Can I use GPIO_USER and ground to connect it? (after assigning it on the console)

I want a to use a GPIO which already has a header pin (not connect direct to the PSF pins)
From the sonoff_template.h - I see the following: 

` GPIO_USER,        // GPIO01 Serial RXD and Optional sensor`
`GPIO_USER,        // GPIO02 Optional sensor`
`GPIO_USER,        // GPIO03 Serial TXD and Optional sensor`

I see the RX and TX headers are available - Is there a GPPIO2 header somewhere on the board?

PSF-85 Wiki page says: 
PIN 9 | GPIO2 | Also used as a programming flash UART1_TX；GPIO2

Not sure where that is...

Thanks for helping a newbie!!!

kajmaj, 2018-03-03:
Newbie to newbie : UART TX is (should be) TX on J6

fanofsonoff, 2018-08-12:
imahmud003 - GPIO2 is SDA. 

I need to connect 2 sensors to Sonoff 4Ch Pro R2. If I connect to RX in J1, it works fine.
But, if I connect sensor to either TX or SDA, sonoff turns-on with 3 orange LEDs and it wont boot. Meaning, I dont see it connected to wifi and nothing works.

However, if I connect sensor to SDA or TX after Sonoff is booted, it works perfectly. I tried with both DC and AC inputs.

I am needing to connect 2 sensors to Sonoff 4CH Pro. Please help on why it wont boot when sensors are connected?