arendst/Tasmota #1104: Shrink code

ionciubotaru, 2017-10-31:
Replace

`void MqttPublishPrefixTopic_P(uint8_t prefix, const char* subtopic, boolean retained)`

with 

`void MqttPublishPrefixTopic_P(uint8_t prefix, const char* subtopic, boolean retained=false)`

then delete

```
void MqttPublishPrefixTopic_P(uint8_t prefix, const char* subtopic)
{
  MqttPublishPrefixTopic_P(prefix, subtopic, false);
}
```

you will save 48 bytes

arendst, 2017-10-31:
Just for fun I tried it and came up with different results.

Arduino IDE 1.8.1 and esp core 2.3.0 +32 bytes more (!?!)
(VisualStudioCode)PlatformIo and esp core 2.3.0 +48 bytes more (!?!)

So please tell me what great compiler you used to get 48 bytes less

ionciubotaru, 2017-11-04:
That's true, sorry, my mistake.

closed 2017-11-04 by ionciubotaru