arendst/Tasmota #1105: Sonoff 4ch restarts

NMGAndy, 2017-10-31:
Sometimes sonoff 4ch restart when receives a mqtt command to switch on one of its relay.
Usually it happens on first start or long standby.

Here is syslog:

Oct 31 15:35:51 sonoff-4ch-1 ESP-WIF: Connected
Oct 31 15:35:51 sonoff-4ch-1 ESP-DNS: Initialized
Oct 31 15:35:51 sonoff-4ch-1 ESP-HTP: Web server active on sonoff-4ch-1.local with IP address 192.168.1.23
Oct 31 15:35:53 sonoff-4ch-1 ESP-MQT: Attempting connection...
Oct 31 15:35:54 sonoff-4ch-1 ESP-APP: (UTC) Tue Oct 31 08:35:53 2017
Oct 31 15:35:54 sonoff-4ch-1 ESP-APP: (DST) Sun Mar 26 02:00:00 2017
Oct 31 15:35:54 sonoff-4ch-1 ESP-APP: (STD) Sun Oct 29 03:00:00 2017
Oct 31 15:35:54 sonoff-4ch-1 ESP-DNS: Query done. MQTT services found 0
Oct 31 15:35:54 sonoff-4ch-1 ESP-MQT: Connected
Oct 31 15:35:54 sonoff-4ch-1 ESP-MQT: tele/sonoff/LWT = Online (retained)
Oct 31 15:35:54 sonoff-4ch-1 ESP-MQT: cmnd/sonoff/POWER =
Oct 31 15:35:55 sonoff-4ch-1 ESP-MQT: tele/sonoff/INFO1 = {"Module":"Sonoff 4CH", "Version":"5.8.0", "FallbackTopic":"DVES_837CD7", "GroupTopic":"sonoffs"}
Oct 31 15:35:55 sonoff-4ch-1 ESP-MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin", "Hostname":"sonoff-4ch-1", "IPAddress":"192.168.1.23"}
Oct 31 15:35:55 sonoff-4ch-1 ESP-MQT: tele/sonoff/INFO3 = {"RestartReason":"Power on"}
Oct 31 15:35:55 sonoff-4ch-1 ESP-RSL: Received Topic domoticz/out, Data Size 282, Data {#012   "Battery" : 83,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "Light/Switch",#012   "id" : "014DC7B5",#012   "idx" : 11,#012   "name" : "Xiaomi Motion Sensor Bedroom1",#012   "nvalue" : 1,#012   "stype" : "Switch",#012   "svalue1" : "0",#012   "switchType" : "Motion Sensor",#012   "unit" : 1#012}
Oct 31 15:35:55 sonoff-4ch-1 ESP-DOM: idx 11, nvalue 1
Oct 31 15:35:55 sonoff-4ch-1 ESP-RSL: Received Topic domoticz/out, Data Size 266, Data {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "Light/Switch",#012   "id" : "00014067",#012   "idx" : 23,#012   "name" : "SN Childroom Switch",#012   "nvalue" : 1,#012   "stype" : "Switch",#012   "svalue1" : "0",#012   "switchType" : "On/Off",#012   "unit" : 1#012}
Oct 31 15:35:55 sonoff-4ch-1 ESP-DOM: idx 23, nvalue 1
Oct 31 15:35:55 sonoff-4ch-1 ESP-DOM: Received Topic /Power1, Data 1
Oct 31 15:35:55 sonoff-4ch-1 ESP-RSL: Group 0, Index 1, Command POWER, Data 1
Oct 31 15:35:55 sonoff-4ch-1 ESP-MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
Oct 31 15:35:55 sonoff-4ch-1 ESP-MQT: stat/sonoff/POWER1 = ON
Oct 31 15:36:00 sonoff-4ch-1 ESP-WIF: Connected
Oct 31 15:36:00 sonoff-4ch-1 ESP-DNS: Initialized
Oct 31 15:36:00 sonoff-4ch-1 ESP-HTP: Web server active on sonoff-4ch-1.local with IP address 192.168.1.23
Oct 31 15:36:02 sonoff-4ch-1 ESP-MQT: Attempting connection...
Oct 31 15:36:02 sonoff-4ch-1 ESP-APP: (UTC) Tue Oct 31 08:36:01 2017
Oct 31 15:36:02 sonoff-4ch-1 ESP-APP: (DST) Sun Mar 26 02:00:00 2017
Oct 31 15:36:02 sonoff-4ch-1 ESP-APP: (STD) Sun Oct 29 03:00:00 2017
Oct 31 15:36:03 sonoff-4ch-1 ESP-DNS: Query done. MQTT services found 0
Oct 31 15:36:03 sonoff-4ch-1 ESP-MQT: Connected
Oct 31 15:36:03 sonoff-4ch-1 ESP-MQT: tele/sonoff/LWT = Online (retained)
Oct 31 15:36:03 sonoff-4ch-1 ESP-MQT: cmnd/sonoff/POWER =
Oct 31 15:36:03 sonoff-4ch-1 ESP-MQT: tele/sonoff/INFO1 = {"Module":"Sonoff 4CH", "Version":"5.8.0", "FallbackTopic":"DVES_837CD7", "GroupTopic":"sonoffs"}
Oct 31 15:36:03 sonoff-4ch-1 ESP-MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin", "Hostname":"sonoff-4ch-1", "IPAddress":"192.168.1.23"}
Oct 31 15:36:03 sonoff-4ch-1 ESP-MQT: tele/sonoff/INFO3 = {"RestartReason":"Power on"}

arendst, 2017-10-31:
You don't happen to feed the 4CH from another sonoff which toggled power to the 4CH as the RestartReason message clearly states it's the result of a power on. Software errors would give other messages.

NMGAndy, 2017-10-31:
Sonoff 4ch connected directly to 220v line. I also noticed that restart reason is Power on. But there is no problem with power.

I use Sonoff 4ch with Domoticz and Xiaomi body sensor. When the sensor switches on - domoticz sends mqqt command and sonoff 4ch switches on (one of the relay).

Ususally it works fine. But few times a day sonoff 4ch restarts on mqtt command - domoticz sends mqtt command to sonoff 4ch - it switches relay on - it is a lamp. I see the light less than second. Then the relay of sonoff 4ch switches off (sonoff 4ch restarts). I loging to sonoff web server and see it is just restarted. But restart reason is power on - very strange.

WollZender, 2017-11-02:
I have a similar problem. Sometimes, after a few MQTT commands to toggle on or off, the Sonoff 4ch restarts. I use the first 3 relays to controll 3 lamps.

The Sonoff is directly connected to 230V.

I found out that the Sonoff works properly if I disconnect the the lamps.

Here is my log:

00:00:00 CFG: Loaded from flash at F7, Count 587
00:00:00 APP: Boot Count 50
00:00:00 Project sonoff Sonoff (Topic OG-FL, Fallback DVES_891864, GroupTopic sonoffs) Version 5.8.0j
00:00:00 CFG: Saved to flash at F6, Count 588, Bytes 1648
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 UPC1CC219B in mode 11N as OG-FL...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on OG-FL.local with IP address 192.168.0.178
00:00:07 MQT: Attempting connection...
00:00:08 DNS: Query done. MQTT services found 0
00:00:08 MQT: Connected
00:00:08 MQT: tele/OG-FL/LWT = Online (retained)
00:00:08 MQT: cmnd/OG-FL/POWER = 
00:00:08 MQT: Subscribe to cmnd/OG-FL/#
00:00:08 MQT: Subscribe to cmnd/sonoffs/#
00:00:08 MQT: Subscribe to cmnd/DVES_891864/#
00:00:08 MQT: tele/OG-FL/INFO1 = {"Module":"Sonoff 4CH", "Version":"5.8.0j", "FallbackTopic":"DVES_891864", "GroupTopic":"sonoffs"}
00:00:08 MQT: tele/OG-FL/INFO2 = {"WebServerMode":"Admin", "Hostname":"OG-FL", "IPAddress":"192.168.0.178"}
00:00:08 MQT: tele/OG-FL/INFO3 = {"RestartReason":"Power on"}
00:00:08 APP: (UTC) Wed Nov 01 17:35:20 2017
00:00:08 APP: (DST) Sun Mar 26 02:00:00 2017
00:00:08 APP: (STD) Sun Oct 29 03:00:00 2017
18:35:22 MQT: stat/OG-FL/RESULT = {"POWER1":"ON"}
18:35:22 MQT: stat/OG-FL/POWER1 = ON
18:35:22 MQT: stat/OG-FL/RESULT = {"POWER2":"OFF"}
18:35:22 MQT: stat/OG-FL/POWER2 = OFF
18:35:22 MQT: stat/OG-FL/RESULT = {"POWER3":"ON"}
18:35:22 MQT: stat/OG-FL/POWER3 = ON
18:35:22 MQT: stat/OG-FL/RESULT = {"POWER4":"OFF"}
18:35:22 MQT: stat/OG-FL/POWER4 = OFF
18:35:29 MQT: tele/OG-FL/STATE = {"Time":"2017-11-01T18:35:29", "Uptime":0, "Vcc":3.156, "POWER1":"ON", "POWER2":"OFF", "POWER3":"ON", "POWER4":"OFF", "Wifi":{"AP":1, "SSId":"UPC1CC219B", "RSSI":94, "APMac":"90:5C:44:6D:6B:05"}}
18:35:29 MQT: tele/OG-FL/SENSOR = {"Time":"2017-11-01T18:35:29", "Switch1":"ON", "Switch2":"OFF", "Switch3":"ON"}
18:35:29 RSL: Group 0, Index 1, Command POWER, Data ON
18:35:29 MQT: stat/OG-FL/RESULT = {"POWER1":"ON"}
18:35:29 MQT: stat/OG-FL/POWER1 = ON
18:35:29 RSL: Group 0, Index 2, Command POWER, Data OFF
18:35:29 MQT: stat/OG-FL/RESULT = {"POWER2":"OFF"}
18:35:29 MQT: stat/OG-FL/POWER2 = OFF
18:35:29 RSL: Group 0, Index 3, Command POWER, Data ON
18:35:29 MQT: stat/OG-FL/RESULT = {"POWER3":"ON"}
18:35:29 MQT: stat/OG-FL/POWER3 = ON

@NMGAndy Did you find out something else?

I'm grateful for any advice.

WollZender

NMGAndy, 2017-11-02:
@WollZender have no news.

I'll try to use another sonoff 4ch module. Also will try newest version of Tasmota. Currently I am using 5.8.

@arendst  do you think it is a hardware problem?

WollZender, 2017-11-06:
I installed 5.9.0 with the same issue.

I connected ground (green-yellow) to the sonoff and behaviour is more stable now, but I can't explain that.

wasi54, 2017-11-06:
Do you have light bulp or halogen lamp connected to the sonoff?
I assume this is hardware! Those kind of lamps have a cold resistance near zero and will shorten your power line for some mili seconds, which then will raise a reset in the sonoff. You may test that by connecting a 10 Ohm resistance in serie to your lamps. This will limit the current to 2 A max. at power on. Or replace your lamps to LED's.

WollZender, 2017-11-07:
@wasi54 
That makes sense. I have one LED lamp, one  compact fluorescent lamp and one halogen lamp. When toggling LED the sonoff seldom restarts.

I will try it with a resistor in serie.

Thank you for the hint.

TommySharpNZ, 2017-11-17:
I noticed this on my 4CH too yesterday.... It's currently just connected to one set of LEDs out on the front porch and sometimes it reboots....
Based on the theories does this mean it's something outside the control of tasmota?

closed 2018-05-07 by stale[bot]