arendst/Tasmota #1106: sonoff touch not connecting to wifi

margusveer, 2017-10-31:
Hi!

I flashed 3 sonoff touch but one of thous don't want to connect to any wifi, I tried different ones but nothing 
logi from serial port something like that 

```
00:00:00 APP: Boot Count 84
00:00:00 Project sonoff �������������������������������������������������������������������������������
�������������������������������������������������������������������������������������������������������
������������������ (Topic �����������������������������������������������������������������������������
��������������������������������������������, Fallback ��������������������������������, GroupTopic ���
����������������������������������������������������) Version 5.8.0q
00:00:01 CFG: Saved to flash at FA, Count 244, Bytes 1648
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:02 WIF: Patch issue 2186
00:00:02 WIF: Connecting to AP255  in mode 11N as �������������������������������...

00:00:08 WIF: Checking connection...
00:00:08 WIF: Attempting connection...
00:00:13 WIF: Checking connection...
00:00:13 WIF: Attempting connection...
00:00:18 WIF: Checking connection...
00:00:18 WIF: Attempting connection...
00:00:23 WIF: Checking connection...
00:00:23 WIF: Attempting connection...
00:00:28 WIF: Checking connection...
00:00:28 WIF: Attempting connection...
00:00:33 WIF: Checking connection...
00:00:33 WIF: Attempting connection...
00:00:38 WIF: Checking connection...
00:00:38 WIF: Attempting connection...
00:00:43 WIF: Checking connection...
00:00:43 WIF: Attempting connection...
00:00:48 WIF: Checking connection...
00:00:48 WIF: Attempting connection...
00:00:53 WIF: Checking connection...
00:00:53 WIF: Attempting connection...
00:00:59 WIF: Checking connection...
00:00:59 WIF: Attempting connection...

0:01:04 WIF: Checking connection...
00:01:04 WIF: Attempting connection...
00:01:07 APP: Button 1 multi-press 1
00:01:09 WIF: Checking connection...
00:01:09 WIF: Attempting connection...
00:01:10 RSL: domoticz/in = {"idx":-1,"nvalue":1,"svalue":""}
00:01:10 RSL: �����������������������������������������������������������������������������������������
���������� = {"POWER":"��������������������������������������������������������������������������������
�������������������������������������������������������������������������������������������������������
�������������������������������������������������������������������������������������������������������
��������������������������������������������������������"}
00:01:10 RSL: �����������������������������������������������������������������������������������������

���������� = ������������������������������������������������������������������������������������������
�������������������������������������������������������������������������������������������������������
�������������������������������������������������������������������������������������������������������
����������������������������������������������
00:01:13 CFG: Saved to flash at F9, Count 245, Bytes 1648
00:01:14 WIF: Checking connection...
00:01:14 WIF: Attempting connection...
00:01:19 WIF: Checking connection...
00:01:19 WIF: Connect failed with AP timeout
00:01:19 WIF: Patch issue 2186
00:01:19 WIF: Connecting to AP256  in mode 11N as �������������������������������...
00:01:23 CFG: Saved to flash at F8, Count 246, Bytes 1648
00:01:24 WIF: Checking connection...
00:01:24 WIF: Attempting connection...
00:01:29 WIF: Checking connection...
00:01:29 WIF: Attempting connection...
00:01:35 WIF: Checking connection...
00:01:35 WIF: Attempting connection...
00:01:40 WIF: Checking connection...
00:01:40 WIF: Attempting connection...
00:01:45 WIF: Checking connection...
00:01:45 WIF: Attempting connection 
```
I even can run the modul in wifi AP mode and when I try change there wifi ssid and password then - 
"Cant find the file" or something like that. 
any help will nice ;)

arendst, 2017-10-31:
You're settings area is corrupt. Try resetting it with command ``reset 2`` from a serial console or web console.

margusveer, 2017-10-31:
@arendst Hi! thanks seems the reset 2 is the magic word - after that at least wifi connected, at home can tell is the MQTT connecting as well ;)

closed 2017-11-01 by arendst