arendst/Tasmota #1107: Connected button to GPIO14 not a switch

freaktech, 2017-10-31:
hi

i want also to use a momentary switch button not a switch with permanent open or close position
at GPIO14

1 short press = stable ON
1 short press  again = OFF

like the original button at the case.

how is this possible

thank you tom

ionciubotaru, 2017-10-31:
read about `switchtopic` and `switchmode`

freaktech, 2017-10-31:
sorry i am new in this, an this will not help

is there a tut for this or more  Infos

i already tried all webinterface Settings

jetema, 2017-10-31:
freaktech  read this link --> https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#mqtt-specific

freaktech, 2017-10-31:
will read it but have no idea what MQTT is   

i think if a can not Change the Settings in weinterface it must be edited in ATOM at the user_config.h and re-flashed?

but not sure what to edit

jetema, 2017-10-31:
ok, forget about mqtt, you can change on web interface, do that, tap on 

CONFIGURATION -> CONFIGURE MODULE  and select  GPIO 14 as a switch

freaktech, 2017-10-31:
@jetema  

sorry you don't understand what i mean

i know this

but i don't want to use a switch !!!

i want to use a   button

jetema, 2017-10-31:
Yes, I see what you want , you want to use a button exactly equal the one is placed on the sonoff's case, so you should activate gpio 14 to work exact equal.
I think via web es easier this task, just type your sonoff's ip on your web browser and click/select  CONFIGURATION -> CONFIGURE MODULE  you'll see all the combinations betwenn gpios and inputs/outputs

freaktech, 2017-10-31:
yes i know this menu

but when i set to  
09 switch 1
or
10 switch 2

its always the same result

https://youtu.be/DAd5l2-0ueQ

jetema, 2017-10-31:
you may try switchmode 1 to 4

freaktech, 2017-10-31:
great its working with

SwitchMode | 3
-- | --

and then in the web interface  09switch1

thx

i have now swedish or so as lang. in the web interface?
any idea how to reset to englisch

davidelang, 2017-10-31:
On Tue, 31 Oct 2017, freaktech wrote:

> will read it but have no idea what MQTT is

MQTT is a protocol for doing IoT automation, it's the main protocol that the 
Tasmota firmware is designed to support (http is an addon)

It is what home automation software such as OpenHab is designed to work with.

jetema, 2017-10-31:
freaktech no idea about the reason of changing language in web browser, try to update to Tasmota 5.9.0 and see what happens

freaktech, 2017-10-31:
i change the language in the fw to german now     

thank you for the help

tom

closed 2018-05-07 by stale[bot]