arendst/Tasmota #1108: Request: Add Support to SUPLA ROW-01

h-tro, 2017-10-31:
Hi,
A few weeks ago, I bought the following device:

SUPLA ROW-01 **by Zamel**
https://supla.zamel.pl/de/produkt/row-01/
The difference to the SUPLA-ESPABLO device, which is already supported by TASMOTA, is that
there are only 1 relay (GPIO05) and 2 sensor inputs (GPIO13 and GPIO14).
Both sensor inputs are designed as 230V AC inputs.

I have successfully flashed the SUPLA ROW-01 with TASMOTA v5.8.0
and configured it as SUPLA-ESPABLO.

With the following change in the file 'sonoff_template.h'
everything seems to work well except for the use of the 230V AC sensor inputs

     "Supla Espablo",	// **changed for Supla ROW-01**
     0,			// GPIO00 Flash jumper
     GPIO_USER,	// GPIO01 Serial RXD and Optional sensor
     GPIO_DSB,	// GPIO02 DS18B20 sensor
     GPIO_USER,	// GPIO03 Serial TXD and Optional sensor
     GPIO_KEY1,	// GPIO04 Button 1
     GPIO_REL1,	// GPIO05 Relay 1 (0 = Off, 1 = On)
     0, 0, 0, 0, 0, 0,	// Flash connection
     0,
     GPIO_USER,        // GPIO13 Optional sensor
     GPIO_USER,        // GPIO14 Optional sensor
     0, 
     GPIO_LED1_INV,    // GPIO16 Led (0 = On, 1 = Off)
     GPIO_ADC0	// ADC0 A0 Analog input

When 230 V AC from an external switch is applied to the sensor inputs of the device, the voltage at GPIO13 or GPIO14 also changes at a frequency of 50 Hz between 3.3 V and GND and causes the relay to flicker. For a 50 Hz cycle of 20 ms, I measured a 10 ms pulse and a 10 ms pause on GPIO14.

1. Can you customize TASMOTA so that if a sensor is configured as a type 'Switch',
Pauses <= 10 ms are not recognized?
The relay should then work stably.
The question is whether pauses <= 10 ms are needed for other use cases when a sensor is configured as a type 'switch'. I do not think so.
Pauses <= 10 ms at the sensor with type 'Switch' can then be ignored in principle.
Otherwise, an additional option would have to be added,
e.g. with an additional 'SetOptionXX' command for 'Sensor Switch' input options.
'SetOptionXX' 0 – default, not taking into account pauses (as before)
'SetOptionXX' 10 - ignores 10 ms pauses on 'Sensor Switch' input (50 Hz AC)
'SetOptionXX' 8.33 - ignores pauses of 8.33 ms at the 'Sensor Switch' input (60 Hz AC, for example, USA)
or
'SetOptionXX' AC Power - ignores all pauses <= 10 ms on 'Sensor Switch' input (50 Hz and 60 Hz AC)

2. Can you then add Supla ROW-01 as a new device in TASMOTA?

      I could test a new developer version.

codmpm, 2018-03-22:
Hey @h-tro, could you give any hint on how to open the SUPLA ROW-01? I don't want to destroy mine while opening it...

Thank you.

h-tro, 2018-06-06:
Hi codmpm,
You can see a small border around the case.
Cut the case around here with a small saw.
But beware, the housing is only 1 mm
Later you can glue it together again.

codmpm, 2018-06-06:
Thank you for the answer... meanwhile I've managed to open it 😉

h-tro, 2018-06-06:
did you flash tasmota?

codmpm, 2018-06-06:
Not yet... the suplay is laying on my desk... time, the usual problem.

h-tro, 2018-07-23:
@arendst 
No support for SUPLA ROW-01 possible ???

ascillato2, 2018-09-04:
Hi,

As we do not have that hardware, please, adapt the code and make a PR.

Thanks.

closed 2018-09-04 by ascillato2