arendst/Tasmota #1109: Multiple TempSensors (DHT22 or DS18b20) on MULTIPLE GPIO's do not work

maxtox, 2017-10-31:
Hi,

just finished investigation about NOT working DS18b20.

I noticed, that if i select f.e. FPIO14=DS18b20 AND GPIO3=ds18b20 then i CAN NOT see any Values.

If i select only ONE GPIO as DS18B20 it works as it shoud!

Could you please advice? I can help with needed LOG etc..

Thank you !

arendst, 2017-10-31:
As designed. Only one GPIO may be configured as DS18B20 as the code just doesn't support more GPIO's.

Also read https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration about support of sensors per group.

maxtox, 2017-10-31:
maybe we can change the option for choosing GPIO's to exclude the same Type of sensor if already choosen?

arendst, 2017-10-31:
Thought about it but that means that on the client side a lot of checks need to be programmed which take up code space to be used for more important features.

maxtox, 2017-10-31:
ok...i see...

just one question...could you point to the line, which says that "Only one GPIO may be configured as DS18B20"...did not find it..

maxtox, 2017-10-31:
i am asking, just to make sure, that another NOOB does not  ask the same again to save you time for "coding etc..."

Sorry!

arendst, 2017-10-31:
Line 9.

maxtox, 2017-10-31:
Only one sensor per group is allowed.

says to me: I can attach ONE sensor for EACH Group. F.e. One ds18b20, onde FHT22, etc...
But as i understood, it is not correct.  It should say:

EACH sensor TYPE can be configured ONLY ON ONE GPIO per Device. NO GPIO's can be configured to the ALREADY EXISTING SENSOR TYPE, o rit should be deleted (set to "NONE") before the Sensor type can be configured to the other GPIO.

What do u think?

maxtox, 2017-10-31:
I can attache multiple DS18b20 to ONE GPIO? or not?

arendst, 2017-10-31:
Yes.

- You can connect multiple DS18b20 to ONLY one GPIO if you enable the driver for multiple DS18x20 in user_config.h.
- You can connect up to Three DHT11 to three SEPARATE GPIOs.

That info is different from the wiki that needs an update on this information.

maxtox, 2017-10-31:
Cool...

thank you ...just found it:

#define USE_DS18x20                              // Optional using OneWire library f**or multiple** DS18B20 and/or DS18S20 (+2k code)

maxtox, 2017-10-31:
ok...you can close it then!

YOU ARE THE VERY BEST!!! If you need any support, please contact me! Just trying to implement K-Type Temp sensor!

closed 2017-10-31 by arendst

userweb2018, 2018-12-08:
Hi, 
I have a wemos d1 mini with Tasmota firmware and 11 ds18b20 sensors connected to 1-wire bus with 4.7 resistor. The problem is that I can read only 8 sensors maximum. This is the limit?

ascillato, 2018-12-08:
Yes

userweb2018, 2019-01-08:
I have the same problem and I wold like to connect more ds18b20 sensors to the wemos d1 mini with Tasmota firmware...
I am not able to rewrite the driver, there is some one that allready done it?
If yes can I have the modified driver?
Thank you for all the good work that you do.
Best regards
Userweb

korneliuszm, 2019-01-08:
How I can read more then 4 Ds18b20 on one gpio? I enabled multiple sensor option, but still it's working only if I connect not more then 4 sensor, when I need to connect 7 of them...
 Best regards

reneb24, 2019-01-10:
Hi @arendst, I tried to follow your advise and figured out that xsns_05_ds18x20.ino is defines a parameter  `DS18X20_MAX_SENSORS  8` .

I updated DS18X20_MAX_SENSORS to the value 16 and built a new firmware.bin. After flashing it to an esp8266 I attached 16 sensors. The Tasmota Web-GUI showed all 16 sensors, but the MQTT payload was cut in the middle of the 16th sensor:

```
...
"DS18B20-15":{"Id":"0316B1C379FF","Temperature":25.7},"DS18B20-16":{"Id":"0316B1C443FF","Temperat
```
When I increase the DS18X20_MAX_SENSORS to 28 and attached 18 sensors, the sensor value list of the Tasmota main page will also be cut, here it breaks in the middle of the 19th sensor.

I guess that the MQTT payload string is limited to any value as well as the sensor value-list on the Tasmota main page. As I do not have any C++ experience I can't figure out where else I have to update the code and if there might be any sideeffects.

Could anybody do the investigation?

austrisv, 2020-01-14:
> I updated DS18X20_MAX_SENSORS to the value 16 and built a new firmware.bin. After flashing it to an esp8266 I attached 16 sensors. The Tasmota Web-GUI showed all 16 sensors, but the MQTT payload was cut in the middle of the 16th sensor

I have a similar experience. I changed MAX to **16**. Although have physically attached 11.  Seeing all 11 on webgui, but mqtt cut at middle of **11**th.
If it changes anything, I also have two PZEM-004t attached/configured to the same esp, reporting the power consumption. 
Tasmota 8.1.0. Sonoff Basic R3 (esp8285)

gampam2000, 2020-02-12:
If i compile it new i get no temperature reading at all, it shows all my 11 sensors and the IDs but no temperature. I even tried it with 8 Sensors, same issue. Did someone had similar problems?