arendst/Tasmota #1110: OTA Update from 5.8.0 to 5.9.0 fails

MatzeJoerling, 2017-10-31:
Hi,

no matter which way i try (Upload via Browser, Google, curl, Firefox) or let tasmota download the bin file, the device reboots with error "Blocked Loop" in the old version.

her is what i have (example device, all others same behavior)

```
Program Version	5.8.0
Build Date & Time	2017-09-18T16:44:29
Core/SDK Version	2_3_0/1.5.3(aec24ac9)
Uptime		6 Hours
Flash write Count	89 at FA000
Boot Count	32
Restart Reason	Blocked Loop
Friendly Name 1	sonoff5

ESP Chip Id		8721692
Flash Chip Id		1327185
Flash Size		1024kB
Program Flash Size	1024kB
Program Size		469kB
Free Program Space	532kB
Free Memory		24kB

```
What i see in syslog / debug to 4:
```
Oct 31 19:17:33 sonoff5 ESP-WIF: Checking connection...
Oct 31 19:17:33 sonoff5 ESP-WIF: Connected
Oct 31 19:17:38 sonoff5 ESP-HTP: Main Menu
Oct 31 19:17:40 sonoff5 ESP-HTP: Firmware Upgrade
Oct 31 19:17:43 sonoff5 ESP-HTP: Upgrade started
Oct 31 19:17:43 sonoff5 ESP-RSL: Received Topic /OtaUrl, Data Size 50, Data http://192.168.2.232/tasmota/sonoff590-minimal.bin
Oct 31 19:17:43 sonoff5 ESP-RSL: Group 0, Index 1, Command OTAURL, Data http://192.168.2.232/tasmota/sonoff590-minimal.bin
Oct 31 19:17:43 sonoff5 ESP-MQT: stat/relais5/RESULT = {"OtaUrl":"http://192.168.2.232/tasmota/sonoff590-minimal.bin"}
Oct 31 19:17:43 sonoff5 ESP-RSL: Received Topic /Upgrade, Data Size 1, Data 1
Oct 31 19:17:43 sonoff5 ESP-RSL: Group 0, Index 1, Command UPGRADE, Data 1
Oct 31 19:17:43 sonoff5 ESP-MQT: stat/relais5/RESULT = {"Upgrade":"Version 5.8.0 from http://192.168.2.232/tasmota/sonoff590-minimal.bin"}
Oct 31 19:17:44 sonoff5 ESP-CFG: Saved to flash at FB, Count 94, Bytes 1648
Oct 31 19:17:46 sonoff5 ESP-HTP: Web server stopped
Oct 31 19:18:26 gateway.hauz24.de dhcpd: DHCPDISCOVER from 60:01:94:85:15:1c (sonoff5) via em0_vlan10
Oct 31 19:18:27 gateway.hauz24.de dhcpd: DHCPOFFER on 192.168.2.132 to 60:01:94:85:15:1c (sonoff5) via em0_vlan10
Oct 31 19:18:27 gateway.hauz24.de dhcpd: DHCPREQUEST for 192.168.2.132 (192.168.2.1) from 60:01:94:85:15:1c (sonoff5) via em0_vlan10
Oct 31 19:18:27 gateway.hauz24.de dhcpd: DHCPACK on 192.168.2.132 to 60:01:94:85:15:1c (sonoff5) via em0_vlan10
```

No matter if i do this with minimal or normal version.

Download URL is valid:
```
$ lynx -head -dump http://192.168.2.232/tasmota/sonoff590-minimal.bin
HTTP/1.1 200 OK
Server: nginx/1.10.3 (Ubuntu)
Date: Tue, 31 Oct 2017 18:29:12 GMT
Content-Type: application/octet-stream
Content-Length: 429328
```

MD5:
883ba87592bcff34f7f91b835a80187e  sonoff590-minimal.bin
343b0a9788b84833f210243c2344e6d9  sonoff590.bin

i have tried with 2 workstations for uploading, i downloaded the sonoff.bin twice ... no idea.

someone else ...?

regards M.

jetema, 2017-10-31:
I tried updating just today from 5.8 and 5.7 to 5.9 and worked good ( from linux & firefox ), so try to update with strong wifi signal 60% or more and if possible into a wifi from the main router, if you try from an AP connected to the main router it happens as you said, Blocked Loop or something similar

tigert, 2017-11-01:
I also got an error when I tested the OTA upgrade.

I pointed the module to https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.9.0/sonoff.bin for the firmware but got this:

11:40:05 HTP: Web server stopped
11:40:06 HTP: Web server active on sonoff1-8162.local with IP address 192.168.1.201
11:40:07 HTP: Web server stopped
11:40:08 HTP: Web server active on sonoff1-8162.local with IP address 192.168.1.201
11:40:09 HTP: Web server stopped
11:40:10 HTP: Web server active on sonoff1-8162.local with IP address 192.168.1.201
11:40:11 HTP: Web server stopped
11:40:12 HTP: Web server active on sonoff1-8162.local with IP address 192.168.1.201
11:40:13 HTP: Web server stopped
11:40:14 HTP: Web server active on sonoff1-8162.local with IP address 192.168.1.201
11:40:16 HTP: Web server stopped
11:40:16 HTP: Web server active on sonoff1-8162.local with IP address 192.168.1.201
11:40:18 HTP: Web server stopped
11:40:18 HTP: Web server active on sonoff1-8162.local with IP address 192.168.1.201
11:40:20 HTP: Web server stopped
11:40:20 HTP: Web server active on sonoff1-8162.local with IP address 192.168.1.201
11:40:22 HTP: Web server stopped
11:40:22 HTP: Web server active on sonoff1-8162.local with IP address 192.168.1.201
11:40:24 HTP: Web server stopped
11:40:24 HTP: Web server active on sonoff1-8162.local with IP address 192.168.1.201
11:40:26 MQT: stat/sonoff1/UPGRADE = Failed HTTP error: connection refused

It also failed on file upload method earlier (I *think* it was running 5.8.0) but I since had used the usb programmer and platformio to update to 5.9.0, and file upload suddenly worked.

11:40:45 MQT: tele/sonoff1/STATE = {"Time":"2017-11-01T11:40:45", "Uptime":0, "Vcc":3.170, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"FresitaS", "RSSI":62, "APMac":"BE:25:E0:8F:BA:44"}}
11:42:33 UPL: File sonoff.bin ...
11:42:45 UPL: Successful 483616 bytes. Restarting.
(this was the successful upgrade)

MatzeJoerling, 2017-11-01:
@tigert it does not work with https. 
because of that use smadds repo ( https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade )

@jetema 
Does that make sense to you?? 
AP1 SSId (RSSI) | IoT-Funk (100%)
Same result: Restart Reason | Blocked Loop

regards M

tigert, 2017-11-02:
I was going to say it happens also with this url (https change to http that also works)

    http://github.com/arendst/Sonoff-Tasmota/releases/download/v5.9.0/sonoff.bin

... but actually it redirects to an Amazon S3 url that serves you the binary. And that one is a HTTPS one.

MatzeJoerling, 2017-11-03:
have you tried: http://sonoff.maddox.co.uk/tasmota/sonoff.bin

smadds, 2017-11-06:
Do use the http copy on my server, @tigert . It gets updated at each release with all of the files released.

There's also an mqtt service that publishes when an update is available...

emontnemery, 2018-01-03:
I have the same problem when wifi signal is poor.
Upload of the firmware easily takes more than 30s, causing the watchdog to bark.
Perhaps the watchdog can be fed in HandleUploadLoop, as long as upload is not stalled?

MatzeJoerling, 2018-01-04:
what means wifi signal poor, and who cares? tcp should handle it! it is designed for poor connections!
we are talking about 350 pakets! 

+1 for better timeout handling!

closed 2018-05-18 by stale[bot]