arendst/Tasmota #1111: Sonoff Basic does not connect to Wifi

Skeeter12, 2017-11-01:
I have tried 3 Sonoff basic but none of them is connecting stable to Wifi. Flashing worked as expected no issues. I also have 3 Pow and they connect to the same network without any problems.

I read here that someone had the same problem but could make it work with older version of the firmware. I tried 4.0, 4.01, 4.03 but all the same result. No connection.

The serial terminal shows this:
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_117E7D, GroupTopic sonoffs) Version 5.8.0
00:00:00 WIF: Connecting to AP1 dritterVersuch in mode 11N as sonoff-7805...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP2 test in mode 11N as sonoff-7805...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WPSConfigactive for 1 minute
00:00:33 APP: Restarting

I tried two different APs (one is called "dritterVersuch" and one "test"). Also I could not connect using WPS. It is the same behaviour for all three of them.
To be sure to be in reach of the APs I placed the Sonoff device just 2 meter away from the AP. Did not help.

Any ideas?

ionciubotaru, 2017-11-02:
Try to flash sonoffs via serial again and set the wifi via serial console, then retry wifi connection

Skeeter12, 2017-11-03:
Thanks for the idea. 

Sadly it did not work. I got connected twice but not stable. I have two more with the stock firmware. I will try them  without flashing to check if the HW is working.

mrohner, 2018-01-26:
Same issue here with a Sonoff Touch. Tasmota 5.11.1f. ESP 2.3.0 

00:00:00 Project sonoff Test1 (Topic sonoff, Fallback DVES_83D9EB, GroupTopic sonoffs) Version 5.11.1f
00:00:00 WIF: Connecting to AP1 mmwireless in mode 11N as sonoff-6635...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 mmwireless in mode 11N as sonoff-6635...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WPSConfigactive for 3 minutes

ionciubotaru, 2018-01-26:
I had similar problems with 2 Sonoffs standard. 
This is not a Tasmota problem is a Sonoff hardware problem - in my case the only choice was to replace the sonoffs.

Skeeter12, 2018-01-29:
Do have any hint on what is wrong with the HW of the Sonoff? Can I somehow see it before buying or before flashing?
I also replaced the Sonoff HW but they seemed to work correctly with the stock FW.

acrollet, 2018-03-01:
I had this issue as well, tried re-flashing several times, could not connect over wi-fi in wifi manager mode. Finally what worked was to install the minimal firmware, which allowed me to connect to the manager, configure wifi connection and then flashed to the OTA firmware.

likaci, 2018-03-02:
@acrollet 
flash the minimal bin then re-flash the normal bin works.
thank you.