arendst/Tasmota #1115: FW 5.9 backup via webinterface - error

freaktech, 2017-11-02:
i tried to save the config

but always get the error

file   config_xxxxx.dmp  could not be downloaded

thx
tom

arendst, 2017-11-02:
Try another browser. Firefox is known not to work.

Tresett, 2017-11-02:
Linux wget not download too:

```
root@raspberrypi1:~# wget http://192.168.5.3/dl
--2017-11-02 16:03:54--  http://192.168.5.3/dl
Connecting to 192.168.5.3:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4096 (4.0K) [application/octet-stream]
Saving to: ‘dl’
dl                 40%[=====>          ]   1.61K  --.-KB/s   in 1.9s
2017-11-02 16:03:56 (865 B/s) - Connection closed at byte 1648. Retrying.

```

arendst, 2017-11-02:
As the config settings are just 1648 bytes long what more to expect...

arendst, 2017-11-02:
Thanks for the hint regarding file size!!

Quickfix: Change line 981 in webserver.ino from
```
  WebServer->setContentLength(4096);
```
into
```
  WebServer->setContentLength(sizeof(SYSCFG));
```

closed 2017-11-07 by arendst