arendst/Tasmota #1117: Touch goes to "wdt reset"

radekmar, 2017-11-02:
Hi All,

First of all - I'm quite new to this home automation stuff. I'm trying to use SonOff components. So far I have managed to use SonOff BASIC using firmware v5.8.0. Yesterday I have received a new toy - SonOff Touch. I've used the same firmware and after a successful upload it didn't work. The touch module appears on my WiFi for few seconds. I can see DHCP lease details and I can connect via web interface but it does not last long. To diagnose the problem I've enabled reporting over RS232 and changed the module type from _BASIC to _TOUCH. Below is what I'm getting back on the RS232 console:

```
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 APP: Button 1 multi-press 1
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff-3005.local with IP address 192.168.1.153
00:00:05 HTP: Main Menu
00:00:06 MQT: Attempting connection...
00:00:06 APP: (UTC) Wed Nov 01 23:36:44 2017
00:00:06 APP: (DST) Sun Mar 26 02:00:00 2017
00:00:06 APP: (STD) Sun Oct 29 03:00:00 2017
00:36:45 DNS: Query done. MQTT services found 0
00:36:45 MQT: Connected
00:36:45 MQT: tele/sonoff/LWT = Online (retained)
00:36:45 MQT: cmnd/sonoff/POWER = 
00:36:46 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Touch", "Version":"5.8.0", "FallbackTopic":"DVES_8CCBBD", "GroupTopic":"sonoffs"}
00:36:46 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin", "Hostname":"sonoff-3005", "IPAddress":"192.168.1.153"}
00:36:46 MQT: tele/sonoff/INFO3 = {"RestartReason":"Power on"}
00:36:47 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
00:36:47 MQT: stat/sonoff/POWER = OFF
00:36:48 RSL: Received Topic /Reset, Data Size 1, Data 1
00:36:48 RSL: Group 0, Index 1, Command RESET, Data 1
00:36:48 MQT: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
00:36:49 CFG: Use defaults
00:36:55 APP: Syslog Host not found. Retry in 600 sec
00:36:56 CFG: Saved to flash at FB, Count 1, Bytes 1648
00:36:57 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

Please let me know were is the problem, how to fix it or diagnose further.

Thank you in advance.

arendst, 2017-11-02:
At 00:00:04 you see a message about Button 1. This means your touch sensor is somehow stuck and keeps the signal low for over 5 seconds which at 00:36:48 performs a reset of the device.

Verify your hardware around GPIO00.

radekmar, 2017-11-02:
It was reporting activity on this occasion as I've left GPIO0 linked to the ground and the log was captured straight after the firmware was uploaded. I've removed this GPIO0 from the ground straight after programming, but clearly was visible in the log.

When I restart the device it does not report anything over RS232 :( it looks like it's dead. The only time it reports something is straight after programming.

arendst, 2017-11-02:
If you compiled the firmware yourself what flash mode option did you use?

It should have been DOUT as documented here: https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips

radekmar, 2017-11-02:
Thank you so much for quick response.

Regarding settings I'm using the following:
[http://tbf.me/a/GVhEw](url)

Board: Generic ESP8285 Module
CPU Frequency: 80MHz
Flash Size: 1M (64K SPIFFS)
Upload Speed: 115200

I'm using the exact same settings on my SONOFF_BASIC and they are working fine.

I've just re-flashed it again - this time I have started the device with pin 5 (GPIO0) linked to the ground and then before programming I've disconnected it from the ground.

```
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:03 APP: Button 1 multi-press 1
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Checking connection...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Patch issue 2186
00:00:07 WIF: Connecting to AP2 test in mode 11N as sonoff-3005...
00:00:07 RSL: Received Topic /Reset, Data Size 1, Data 1
00:00:07 RSL: Group 0, Index 1, Command RESET, Data 1
00:00:07 RSL: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
00:00:08 CFG: Saved to flash at F9, Count 3, Bytes 1648
00:00:08 CFG: Use defaults
00:00:09 CFG: Saved to flash at FB, Count 1, Bytes 1648
00:00:09 WIF: Checking connection...
00:00:09 WIF: Attempting connection...
00:00:10 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

```

radekmar, 2017-11-02:
Sorry to be a pain -> DOUT setting is available only on Module ESP8266. I've recompiled it using those settings (link below)

[http://tbf.me/a/BdOd8U](url)

and it still goes through the same route of "wdt reset"

```
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:03 APP: Button 1 multi-press 1
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Checking connection...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Patch issue 2186
00:00:07 WIF: Connecting to AP2 test in mode 11N as sonoff-3005...
00:00:07 RSL: Received Topic /Reset, Data Size 1, Data 1
00:00:07 RSL: Group 0, Index 1, Command RESET, Data 1
00:00:07 RSL: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
00:00:08 CFG: Saved to flash at F9, Count 3, Bytes 1648
00:00:08 CFG: Use defaults
00:00:09 CFG: Saved to flash at FB, Count 1, Bytes 1648
00:00:09 WIF: Checking connection...
00:00:09 WIF: Attempting connection...
00:00:10 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

```

arendst, 2017-11-02:
After initial firmware load it will always go to wdt reset!

You'll have to power cycle the device to get the initial firmware going.

radekmar, 2017-11-02:
That's the problem - the module does not start after power cycle, does not send any info over RS232 and does not connect to the WiFi. :(

stefanbode, 2017-11-02:
As @arendst already pointed out there is an error with your PIN. Seet at:
00:00:03 APP: Button 1 multi-press 1
This should not be there on normal reboot. And this is the cause it reboots later on. You have 7 seconds to go to type the following command. Please try a "reset 2". Maybe there is some "dirt" in the configuration flash. The reset 2 will erase the flash before reboot and start with a definite clean one.

radekmar, 2017-11-02:
Thank you very much.

The result of "reset 2" is listed below. Unfortunately the problem persists - after the power cycle the board does not start (no RS232 data nor WiFi activity).

```
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 CMD: reset 2
00:00:03 RSL: Received Topic /reset, Data Size 1, Data 2
00:00:03 RSL: Group 0, Index 1, Command RESET, Data 2
00:00:03 RSL: stat/sonoff/RESULT = {"Reset":"Erase, Reset and Restarting"}
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:03 APP: Button 1 multi-press 1
00:00:04 APP: Erase 143 sectors
APP: Erased sector 113 Ok
APP: Erased sector 114 Ok
APP: Erased sector 115 Ok
APP: Erased sector 116 Ok
APP: Erased sector 117 Ok
APP: Erased sector 118 Ok
APP: Erased sector 119 Ok
APP: Erased sector 120 Ok
APP: Erased sector 121 Ok
APP: Erased sector 122 Ok
APP: Erased sector 123 Ok
APP: Erased sector 124 Ok
APP: Erased sector 125 Ok
APP: Erased sector 126 Ok
APP: Erased sector 127 Ok
APP: Erased sector 128 Ok
APP: Erased sector 129 Ok
APP: Erased sector 130 Ok
APP: Erased sector 131 Ok
APP: Erased sector 132 Ok
APP: Erased sector 133 Ok
APP: Erased sector 134 Ok
APP: Erased sector 135 Ok
APP: Erased sector 136 Ok
APP: Erased sector 137 Ok
APP: Erased sector 138 Ok
APP: Erased sector 139 Ok
APP: Erased sector 140 Ok
APP: Erased sector 141 Ok
APP: Erased sector 142 Ok
APP: Erased sector 143 Ok
APP: Erased sector 144 Ok
APP: Erased sector 145 Ok
APP: Erased sector 146 Ok
APP: Erased sector 147 Ok
APP: Erased sector 148 Ok
APP: Erased sector 149 Ok
APP: Erased sector 150 Ok
APP: Erased sector 151 Ok
APP: Erased sector 152 Ok
APP: Erased sector 153 Ok
APP: Erased sector 154 Ok
APP: Erased sector 155 Ok
APP: Erased sector 156 Ok
APP: Erased sector 157 Ok
APP: Erased sector 158 Ok
APP: Erased sector 159 Ok
APP: Erased sector 160 Ok
APP: Erased sector 161 Ok
APP: Erased sector 162 Ok
APP: Erased sector 163 Ok
APP: Erased sector 164 Ok
APP: Erased sector 165 Ok
APP: Erased sector 166 Ok
APP: Erased sector 167 Ok
APP: Erased sector 168 Ok
APP: Erased sector 169 Ok
APP: Erased sector 170 Ok
APP: Erased sector 171 Ok
APP: Erased sector 172 Ok
APP: Erased sector 173 Ok
APP: Erased sector 174 Ok
APP: Erased sector 175 Ok
APP: Erased sector 176 Ok
APP: Erased sector 177 Ok
APP: Erased sector 178 Ok
APP: Erased sector 179 Ok
APP: Erased sector 180 Ok
APP: Erased sector 181 Ok
APP: Erased sector 182 Ok
APP: Erased sector 183 Ok
APP: Erased sector 184 Ok
APP: Erased sector 185 Ok
APP: Erased sector 186 Ok
APP: Erased sector 187 Ok
APP: Erased sector 188 Ok
APP: Erased sector 189 Ok
APP: Erased sector 190 Ok
APP: Erased sector 191 Ok
APP: Erased sector 192 Ok
APP: Erased sector 193 Ok
APP: Erased sector 194 Ok
APP: Erased sector 195 Ok
APP: Erased sector 196 Ok
APP: Erased sector 197 Ok
APP: Erased sector 198 Ok
APP: Erased sector 199 Ok
APP: Erased sector 200 Ok
APP: Erased sector 201 Ok
APP: Erased sector 202 Ok
APP: Erased sector 203 Ok
APP: Erased sector 204 Ok
APP: Erased sector 205 Ok
APP: Erased sector 206 Ok
APP: Erased sector 207 Ok
APP: Erased sector 208 Ok
APP: Erased sector 209 Ok
APP: Erased sector 210 Ok
APP: Erased sector 211 Ok
APP: Erased sector 212 Ok
APP: Erased sector 213 Ok
APP: Erased sector 214 Ok
APP: Erased sector 215 Ok
APP: Erased sector 216 Ok
APP: Erased sector 217 Ok
APP: Erased sector 218 Ok
APP: Erased sector 219 Ok
APP: Erased sector 220 Ok
APP: Erased sector 221 Ok
APP: Erased sector 222 Ok
APP: Erased sector 223 Ok
APP: Erased sector 224 Ok
APP: Erased sector 225 Ok
APP: Erased sector 226 Ok
APP: Erased sector 227 Ok
APP: Erased sector 228 Ok
APP: Erased sector 229 Ok
APP: Erased sector 230 Ok
APP: Erased sector 231 Ok
APP: Erased sector 232 Ok
APP: Erased sector 233 Ok
APP: Erased sector 234 Ok
APP: Erased sector 235 Ok
APP: Erased sector 236 Ok
APP: Erased sector 237 Ok
APP: Erased sector 238 Ok
APP: Erased sector 239 Ok
APP: Erased sector 240 Ok
APP: Erased sector 241 Ok
APP: Erased sector 242 Ok
APP: Erased sector 243 Ok
APP: Erased sector 244 Ok
APP: Erased sector 245 Ok
APP: Erased sector 246 Ok
APP: Erased sector 247 Ok
APP: Erased sector 248 Ok
APP: Erased sector 249 Ok
APP: Erased sector 250 Ok
APP: Erased sector 251 Ok
APP: Erased sector 252 Ok
APP: Erased sector 253 Ok
APP: Erased sector 254 Ok
APP: Erased sector 255 Ok
00:00:15 CFG: Use defaults
00:00:16 CFG: Saved to flash at FB, Count 1, Bytes 1648
00:00:16 WIF: Checking connection...
00:00:16 WIF: Connect failed as AP cannot be reached
00:00:16 WIF: Patch issue 2186
00:00:16 WIF: Connecting to AP2 test in mode 11N as sonoff-3005...
00:00:17 CFG: Saved to flash at FA, Count 2, Bytes 1648
00:00:17 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

```

stefanbode, 2017-11-02:
From my perspective, your GPIO0 is still connected to ground. This might give you the "view" that it not start. Can you check your soldering at the pins. Maybe just too much solder. As long as GPIO0 is connected to ground the ESP will start in FLASH mode. This enables flashing, but disable the normal start. If it just reboot after start your switch permanently pressed. Check with a resistance measure tool, if your switch is working correctly.

closed 2018-05-07 by stale[bot]