arendst/Tasmota #1118: cmnd/sonoff/Pwm returns {"Command":"Unknown"}

MikePRoberts, 2017-11-02:
I would like to drive 3 pwms directly.  However all cmnd/sonoff/Pwm topics get a {"Command":"Unknown"} response. cmnd/sonoff/PwmFrequency and PwmRange work fine.  I have the three pwms set up on gpios 4,2,15 and can drive them using Color with SetOption15 On.  I have turned SetOption15 Off tried PWM, PWM2, pwm, Pwm all with the same outcome.  Am using firmware 5.8.0p and Module 18 (WeMos D1 mini) - actually an ESP-12F.  Also have a WS2812 string connected on gpio13 (working fine).  Am I missing something?  Thank you.  Mike

arendst, 2017-11-02:
What value did enter after PWM?

I have PWMRange set to 1023 and PWMFrequency set to 880. If I perform command ``pwm1 200`` I get a red color (as that's where the red leds are connected to).

arendst, 2017-11-02:
Ah hold on. Connecting both WS2812 AND PWM controlled leds to the same device is not supported.

MikePRoberts, 2017-11-02:
Thank you.  For what its worth I was able to control one or the other via the color command (depending on SetOption15).  I was just trying to see how many devices I could drive from one ESP-12F!!  Very impressed nevertheless.  Mike

closed 2017-11-02 by MikePRoberts