arendst/Tasmota #1119: Need help for Sonoff Touch T1 3gang version finding pins

kevinshane, 2017-11-02:
Hi guys, pretty noob here, today I've got my Touch T1 UK 3gang version(or I should say T3?) from itead
It looks really different and complicated comparing to the Sonoff Touch and other sonoff like device, would u guys please help me finding the soldering pins for rx/tx/gnd/vcc?
Here is the front side (Q1: what's middle button for? Q2: what's the left and the right side hole for?)
[image]
here is the back side:
[image]
Looks like the J1 from the back side will be the programming pins(I just take a guess here but I have no idea of which pin is which), also should I need to press the middle button(from the front side) when doing the flashing stuff? 
Please help, I can't find any ref images/photos on internet, and I can't get the sonOTA works, the only way is by soldering... Thank you so much!!

Shane

ionciubotaru, 2017-11-03:
The square pin in 3V3, near RX, then TX, and the last one is GND
Better look at the ESP chip's pins and where the connections goes on the PCB or use an ohmmeter.

kevinshane, 2017-11-04:
hi @ionciubotaru thank you so much for your reply, but what square pin from what side? Did u mean the J1(look at from the back side jpg)?
Do u have any idea of what's the meaning of the J3( the one with 5pins hole)
and sorry about the meter thing, I have 0 exp of tracking those pins, I do have a meter but I have no idea of how to trace them
I will give it a try with your idea(according to the J1 square pin is 3v3) and soldering some jumpers, I will report it back for you
and thank you again for your info ^^...have a nice day sir

ionciubotaru, 2017-11-04:
On J1 the square pin should be VCC - try to connect 3V3 and GND to J1 and check if at least one led is ON

kevinshane, 2017-11-04:
hi @ionciubotaru, I couldn't find the gpio0 for the tiny esp8285 (I googled, some said the gpio0 is the pin15 but when I connected the pin15 to gnd, it won't work, not sure why), so I end up with flashing v5.8.0 tasmota by using sonOTA, so far it's working like a charm ^^..

rlust, 2017-11-06:
How did you flash it with out connecting it to USB and grounding Pin 15?

purpleogre, 2017-11-06:
Hello Kevinshane,
I got one of the 2Ch switches just before the weekend.
JP3 has the rx and tx pins on.
pin 1 (square pad) is 3v3, TX,RX,GND and pin5 is a GPio pin, I can't remember which one..but not GPio0
GPio0 in connected to R19 where the silk screen label is.
I managed to flash mine but it keeps loosing connection to the WiFi, and I haven't had time to diagnose it yet although it did surprise me when it beeped, s1 seems to be reset.

purpleogre, 2017-11-06:
Hello Kevinshane
Just looking at your image to refresh my memory pin 5 looks like its to gpoi2 as the track goes up to and then in between R18 and R19

kevinshane, 2017-11-08:
hi @purpleogre, sorry I don't have a 2ch sonoff touch, but I assume it's the same as 3gang version
I end up using this to flash the 3gang version, so far it's working like a charm
https://github.com/arendst/Sonoff-Tasmota/wiki/SonOTA
by using SonOTA, u don't need to know which pin is which or even no need to open the cover, you just... flash it... without any soldering

Bockwurst19, 2017-11-11:
Hi @all
Also have a Sonoff T1 Touch.
Could show me where 3v3 RX TX GND and GPIO 0 is on the board.

Thanks MFG

Wieg, 2017-11-11:
i am exactly at the same moment .... my sonoff's just came in 30 minutes ago and i was expecting another type ..... now i need to flash this exactly this type .... any help would be really appreciated

purpleogre, 2017-11-11:
[image]

Wieg, 2017-11-11:
thanks man butttuuuhhh ? what is my 3.3v and the ground ? sorry

Wieg, 2017-11-11:
ok ... gonna flash in a few minutes .... wish my luck

Wieg, 2017-11-11:
i am not getting it in flashmode ..... :( ideas ???

Wieg, 2017-11-11:
SonOTA regretfully isn't working for this 2 ..... @Bockwurst19 di you have any luck ?

Wieg, 2017-11-11:
I got the new firmware in ..... the 5.9.1c ..... but now the device in not booting ... any ideas ? i did the sonoff firmware and only changed the wifi setting userconfig. I did not change the 8266 in 8265 ..... but i think platformio does that for me ????

purpleogre, 2017-11-11:
Sorry for the late reply I've been out to see Thor Ragnorak. 
I presume you figured out 3v3 is 3.3V,and that you shorted gpio0 to 0V to get into flash mode.
I also use platformio, I presume that you remembered to uncomment
env_default = sonoff in platformio.ini file (or the one for your language)
and in  user_config.h I always change the details for WIFI_GATEWAY,WIFI_SUBNETMASK,WIFI_DNS,STA_SSID1 and STA_PASS1 to connect to my network.
since I use Homeassistant I also change the MQTT_HOST and associated port, user and password and set MQTT_POWER_RETAIN to 1
I also change settings for the NTP servers,  and Time setting to reflect that i'm in the UK.
Once compiled and flashed I will go onto my router see what it's details are and then assign a static ipaddress, reboot it and then use the static address to use the web interface to change to the correct module type etc..

Wieg, 2017-11-11:
dude sorry ????? you're great to reply thank you !!
 i managed to flash one .... and brick one i think .... 

https://github.com/arendst/Sonoff-Tasmota/issues/1170

any ideas always welcome

2e0vet, 2017-11-20:
purpleogre  can you check your image, I think you have TX/RX the wrong way round (unless that was how to connect to TTL)

purpleogre, 2017-11-23:
@2e0vet modified to make it clearer :-)

kevinshane, 2017-11-27:
@purpleogre awesome man!! thanks for sharing with us, may I ask is the 2gang version exactly the same this one(3gang version), this is because I have the sonoff touch 1gang version, and it's totally different world comparing to this one

josemacaluso, 2017-12-11:
@purpleogre Hi, I'm trying to flash a T1 2gang with no success. Try to use the latest versions of sonoff-tasmota (5.10.0) and only change the board=8285, but I have ever memory size issues (The program size (476957 bytes) is greater than maximum allowed (434160 bytes)), do you already do it, may you share the platformio.ini and user_config.h files you used to reach the goal? Thanks

jimmyw, 2018-01-06:
I missed this thread, i started a wiki page here:

https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1

Please help me add your findings there!

kylegordon, 2018-01-18:
@kevinshane The holes on the left and right are for the screws to go through :-)

mattpurnell, 2018-01-25:
@purpleogre firstly thanks a million for getting the info out there. But - how did you ground the impossibly small GPio0 PIN!?

closed 2018-06-23 by stale[bot]