arendst/Tasmota #1121: Sonoff pow Not working after update

mittelfeldmotor, 2017-11-02:
I updated an sonoff pow from 5.4.0 to 5.9.0, now only the WiFi LED works. No relay,no webif, nothing.
Anyone an idea or should i flash with fdti again?

ionciubotaru, 2017-11-03:
try to reflash it via serial

stefanbode, 2017-11-03:
I had the same issue with 5.8.0q and the POW. After the OTA update, the device did not reboot again. Also, a serial update was a bit tricky. Holding the key just during power up did not bring the device in flash mode. Flashing stops with an error. At the end, it worked with HOLDING the button all time until flashing was done. Now the device is up and running.

mittelfeldmotor, 2017-11-05:
thanks stefan, I flashed it like you sugguesteted, and it works now.

closed 2017-11-05 by mittelfeldmotor