arendst/Tasmota #1122: Not enough space to flash?

mB-PiBox, 2017-11-02:
Archiving built core (caching) in: C:\Users\marsh\AppData\Local\Temp\arduino_cache_492097\core\core_esp8266_esp8266_generic_UploadTool_esptool,CpuFrequency_80,FlashFreq_40,FlashMode_dout,UploadSpeed_115200,FlashSize_1M0,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____37bed59924cb581baf64a62f6a880f83.a

Sketch uses 481216 bytes (46%) of program storage space. Maximum is 1023984 bytes.
Global variables use 45388 bytes (55%) of dynamic memory, leaving 36532 bytes for local variables. Maximum is 81920 bytes.

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

zerog2k, 2017-11-07:
Do you have it set to wemo or hue emulation?

I am not able to get my echo pro to discover my tasmota device in wemo emulation, but works with hue emulation with a caveat: turning on device from echo results in the device turning on but alexa saying something about not responding. When I turn off the device through echo this works as expected with Alexa responding with OK.

mB-PiBox, 2017-11-07:
1. There is no "Echo Pro."
2. I have them all set to WeMo

zerog2k, 2017-11-07:
Sorry it was late when posting. It’s echo plus.

Alveran81, 2017-11-15:
I have a Echo Plus and Echo Dot. My Echo Plus can not find or switch wemo device. When I switch Echo Plus off my echo dot can find and switch wemo devices.

zerog2k, 2018-04-23:
Not sure at what point this was fixed, but it seems to be reliably working for me for past few months... Something may have been fixed at some point

closed 2018-06-22 by stale[bot]