arendst/Tasmota #1128: Feature: Xenon 3 gang US Light Switch

kezzab25, 2017-11-05:
I have finally got the Xenon 3CH WIFI US switch working on Tasmota. If someone can add it to the code that would be awesome. They are the US size, not the UK which is why I wanted it to work. 

`{ "Xenon 3CH",      // Xenon 3CH (ESP8266)
    0,0,0,
    GPIO_KEY2,        // GPIO03 Serial TXD and Optional sensor
    GPIO_REL2,        // GPIO04 Relay 2
    GPIO_KEY3,        // GPIO05 Input 2
    0, 0, 0,          // Flash connection
    0,
    0,
    0,
    GPIO_KEY1,        // GPIO12 Key input
    GPIO_REL1,        // GPIO13 Relay 1
    0,
    GPIO_REL3,        // GPIO14 Relay 3
    0, 0
  }`

I also had to add XENON3 after WITTY to make it work.

davidelang, 2017-11-05:
Is there a reason you can't just use the generic wemo and  define the 
appropriate GPIOs?

On Sun, 5 Nov 2017, Kieren wrote:

> Date: Sun, 05 Nov 2017 01:38:19 +0000 (UTC)
> From: Kieren <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f2a4659dfb9a0d305a6a196eda9092e83776d28f792cf0000000116162d8
>     a92a169ce102ace64@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: [arendst/Sonoff-Tasmota] Feature: Xenon 3 gang US Light Switch
>     (#1128)
> 
> I have finally got the Xenon 3CH WIFI US switch working on Tasmota. If someone can add it to the code that would be awesome. They are the US size, not the UK which is why I wanted it to work.
>
> `{ "Xenon 3CH",      // Xenon 3CH (ESP8266)
>    0,0,0,
>    GPIO_KEY2,        // GPIO03 Serial TXD and Optional sensor
>    GPIO_REL2,        // GPIO04 Relay 2
>    GPIO_KEY3,        // GPIO05 Input 2
>    0, 0, 0,          // Flash connection
>    0,
>    0,
>    0,
>    GPIO_KEY1,        // GPIO12 Key input
>    GPIO_REL1,        // GPIO13 Relay 1
>    0,
>    GPIO_REL3,        // GPIO14 Relay 3
>    0, 0
>  }`
>
> I also had to add XENON3 after WITTY to make it work.
>
>

kezzab25, 2017-11-05:
I could but would it not be good to have there so that others can use it
without having to try and work out the GPIO's?

davidelang, 2017-11-05:
only if it's common enough, we are short of space and so having definitions for 
every device out there is unaffordable.

document what's needed on the wiki, complete with pictures so that other people 
can use it easily

CZ1979, 2017-12-19:
@kezzab25 : Im really interested in how you flashed your Xenon switch! Which pins did you used? Which tool and which configuration? Maybe you have a picture of your device?

webtop, 2017-12-19:
@kezzab25 I also would be interested in seeing a Wiki page on how you did this.

kezzab25, 2017-12-19:
I will post some images and how I did everything this weekend.
Unfortunately I am busy at work during the week.

hahmann, 2018-05-03:
I'm curious as well on how you did it @kezzab25 . Havent tried to remove the ESP12F module though, it seems the easiest option to me now. Will be really awesome if you can post some pictures on how you manage to wire it and upload the code. Thanks.

kezzab25, 2018-05-03:
I would have to dismantle it which is a bit more difficult now. I soldered
wires directly onto the esp module and did it that way. I noticed sonoff
have just released a 3 gang light switch for the US so that might be a
better option

hahmann, 2018-05-03:
Having wires soldered into the board directly is not working, some of the pins fluctuate and the espcomm_sync/open/upload failed, maybe due to the other components on the board. Serial is working as in normal operation I see messages coming on it. Did you had the same issue?

kezzab25, 2018-06-17:
Did you remember to hold GPIO0 low on power up?

closed 2018-08-16 by stale[bot]