arendst/Tasmota #1129: Want to use GPIO 3 (RXD) on Sonoff Dual

WollZender, 2017-11-05:
Hello,

I hope somebody can help me.

I try to use GPIO3 (RXD) and GPIO13 (blue LED) as Input for two wall Switches. I don't want using GPIO 4/14, because it is difficult to connect a switch there. I removed the blue LED an connect a Switch to GPIO13. This works properly, but with a switch at GPIO3 nothing happens when I toggle the switch. I already disabled serial logging.

I know that the relays will toggled by the TXD (GPIO1) pin.

Somebody knows what I'm doing wrong?

many thanks

WollZender

davidelang, 2017-11-05:
you cannot use GPIO3 on the dual because that is how the dual controls the 
relays, it's not just the tx pin.

On Sun, 5 Nov 2017, WollZender wrote:

> Date: Sun, 05 Nov 2017 11:13:54 +0000 (UTC)
> From: WollZender <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24fc1adb54fe2b2dc6cd17fe6e3cfaddb8641c8f69092cf000000011616b47
>     292a169ce102b395b@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: [arendst/Sonoff-Tasmota] Want to use GPIO 3 (RXD) on Sonoff Dual
>     (#1129)
> 
> Hello,
>
> I hope somebody can help me.
>
> I try to use GPIO3 (RXD) and GPIO13 (blue LED) as Input for two wall Switches. I don't want using GPIO 4/14, because it is difficult to connect a switch there. I removed the blue LED an connect a Switch to GPIO13. This works properly, but with a switch at GPIO3 nothing happens when I toggle the switch. I already disabled serial logging.
>
> I know that the relays will toggled by the TXD (GPIO1) pin.
>
> Somebody knows what I'm doing wrong?
>
> many thanks
>
> WollZender
>
>

WollZender, 2017-11-05:
I also disconnected the RX between its pin and resistor R10 and it still works. The RX pin now is only connected to GPIO3. Is it possible to use only GPIO1 for sending?

davidelang, 2017-11-05:
not with the stock serial library, it claims both GPIOs

WollZender, 2017-11-05:
Is it possible to tell the library e.g. GPIO4 as serial Input?

Thanks for your reply.

davidelang, 2017-11-05:
no, the hardware has a single serial port, you can't use any other ports for the 
serial port.

WollZender, 2017-11-06:
Thank you for pointing me in the right direction.

I now just open the serial port to send the relay command. After that I close the port to use GPIO3 as Input.

This works properly.

closed 2018-05-07 by stale[bot]