arendst/Tasmota #1130: Feature request: CRC32 check on configuration loaded

stefanbode, 2017-11-05:
The recent changes and the OTA updates showed that based on some strange things the device did not come up properly with the configuration found. I know it should work, but it did only 80% of the time. Especially when upgrading from a 5.0 or similar older version. Changing the configuration back to default at least make the device accessible again. Therefore my request: ad a u_int32 or similar to do a CRC check on the loaded config; just simple, no fancy stuff. If passed you load, if not start with a default config. This would help me not to disassemble the devices after a failed upgrade. Currently, I lose control and have manual serial update them with a temporary change on the CFG_HOLDER.

davidelang, 2017-11-05:
how would the CRC check be any better than the CFG_HOLDER for checking if a 
config is valid or not?

The actions you propose after the check are the same ones performed today if the 
CFG_HOLDER check doesn't work.

stefanbode, 2017-11-06:
As oyu know I have some additions in my code that are also in the CONFIG file. Therefore is can happen that the CFG_HOLDER ID is ok ant match, but because of the changes the CONFIG itself is not valid. The ID does not ensure the data in the BIN is ok to boot.

davidelang, 2017-11-06:
it doesn't matter what changes you make in the config file, the config in the 
flash can be what you consider 'invalid'

it doesn't matter if you look for a special value or a special checksum of the 
data, it still isn't going to match what you put in the config. It's slightly 
less likely to match a checksum than it is to match the magic value of 
GFG_HOLDER, but both are very unlikely.

you are far more likely to have a config that doesn't work for your hardware, 
but is a valid sonoff config than you are to have something that's invalid for 
everything, but still has the CFG_HOLDER value there.

stefanbode, 2017-11-07:
Hi David, the problem only occurs on UPDATING devices with an existing and running CONFIG to a new version. All other use-cases are fine. Arendt did some spare space in the config holder and now track on a bit-base what to change and where to change (he is already aware of the same problem I figured out). For me it looks like the upgrade process from one version to the other does not work reliably, if there is more than one 0.x version in between. Lets close this request. Maybe it is already better right now with the bitwise check that arend do when making changes to the config.

closed 2017-11-07 by stefanbode