arendst/Tasmota #1131: Unauthenticated wireless password disclosure

strasharo, 2017-11-05:
Hello. I'm running version 5.9.0 on a Sonoff S20 relay and have configured a password for the web interface. If for some reason it loses connectivity to the wireless access point (either due to the AP not being present or if I forcefully de-authenticate it from the AP using packet injection) the relay reboots and starts broadcasting an unencrypted wireless network with the relay name as a SSID. When I connect to that wireless network and open http://192.168.4.1/ I'm presented with the Wifi configuration screen without being asked for any authentication even though I've set a password for the web interface. From there I can just use "Inspect element" and reveal the configured wireless password in clear text. This way anyone within the wireless range of the relay can force de-authenticate it from the network and connect to it and retrieve the wireless password. It would be good if the specific screen requires authentication if it was configured for the web interface.

davidelang, 2017-11-05:
change the wifi config, it has the ability to do multiple thingsat startup, and 
if the first connection fails.

strasharo, 2017-11-05:
Thanks a lot, set WifiConfig to 5 and now I does what I want. :)

closed 2017-11-05 by strasharo

chaoranxie, 2018-07-22:
@strasharo i am having the exact same issue and i am not sure "WifiConfig 5" solves the issue. so if your house lost power, or you unplug the S20, the setting goes back to "WifiConfig 2".

Can you check if this is the case for you? it is the case for me

strasharo, 2018-07-22:
@chaoranxie just retested with the latest firmware, it's persistent for me. Unplugged it for 10 minutes and plugged it back it and it was still there.

meingraham, 2018-07-22:
Where (which header file) does one set WifConfig?

strasharo, 2018-07-22:
I'm setting it through the web console once and it persists on the device. Regarding the header file, it's set here:
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/user_config.h#L66
Values are documented here in the wifi section:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

meingraham, 2018-07-22:
@strasharo I suspected that was it, but the slight name difference and the fact that it was being referred to as a value rather than a symbol... I just wanted to ensure that this wasn't something altogether different. Thanks.