arendst/Tasmota #1132: Sonoff Basic Power on Power Off

mittelfeldmotor, 2017-11-05:
I have an basic with 5.8.0 
GPIO14 is an switch
16:50:43 MQT: stat/sonoff-A00F7D/RESULT = {"POWER":"OFF"}
16:50:43 MQT: stat/sonoff-A00F7D/POWER = OFF
16:50:43 MQT: stat/sonoff-A00F7D/RESULT = {"POWER":"ON"}
16:50:43 MQT: stat/sonoff-A00F7D/POWER = ON
16:52:33 MQT: stat/sonoff-A00F7D/RESULT = {"POWER":"OFF"}
16:52:33 MQT: stat/sonoff-A00F7D/POWER = OFF
16:52:34 MQT: stat/sonoff-A00F7D/RESULT = {"POWER":"ON"}
16:52:34 MQT: stat/sonoff-A00F7D/POWER = ON

I do nothing and this happens every few minutes.
Any idea what this is

edit21, 2017-11-05:
Hi
Sounds like it subscribed to something on the mqtt broker, whats controlling commands onto your broker?

mittelfeldmotor, 2017-11-05:
I have 3 others here with the same setup,but withot gpio14 switch,and they work perfect. I am confussed

davidelang, 2017-11-05:
enable more detailed logging,that will tell you where the commands are coming 
from.
k

davidelang, 2017-11-05:
Ok, in that case it's possible you need a pull-up resistor on GPIO14, see the 
troubleshooting page on the wiki about this.

mittelfeldmotor, 2017-11-05:
ok, I ordered some an try it out.

ionciubotaru, 2017-11-06:
It is not a transistor - it is a resistor betwee GPIO and 3V3.
If you still have problems add a capacitor (a small one) between GPIO and GND

Himbeere31415926, 2017-11-11:
@mittelfeldmotor ... did you solve it? i've got the same issue.

closed 2018-05-07 by stale[bot]