arendst/Tasmota #1133: H801 module restart for WakeupDuration command

dimitrystd, 2017-11-05:
Just got `h801` module and flashed it with the last available 5.9.0a. Also set `SetOption15 1`.
When i send command `WakeupDuration 1` then it restarts with error
```
00:00:00 Project h801_bedroom Sonoff (Topic h801_bedroom, Fallback h801_bedroom_422384, GroupTopic sonoffs) Version 5.9.0a
00:00:00 WIF: Connecting to AP1 600-1 in mode 11N as h801_bedroom-0900...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on h801_bedroom-0900 with IP address 192.168.1.62
11:44:24 MQT: Attempting connection...
11:44:24 MQT: Connected
11:44:24 MQT: tele/h801_bedroom/LWT = Online (retained)
11:44:24 MQT: cmnd/h801_bedroom/POWER = 
11:44:24 MQT: tele/h801_bedroom/INFO1 = {"Module":"H801", "Version":"5.9.0a", "FallbackTopic":"h801_bedroom_422384", "GroupTopic":"sonoffs"}
11:44:24 MQT: tele/h801_bedroom/INFO2 = {"WebServerMode":"Admin", "Hostname":"h801_bedroom-0900", "IPAddress":"192.168.1.62"}
11:44:24 MQT: tele/h801_bedroom/INFO3 = {"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x4000bf64 epc2:0x00000000 epc3:0x40204483 excvaddr:0x00000001 depc:0x00000000"}
```
If i send just `WakeupDuration` then console stops to work. I refresh page in browser but it doesn't help. Only restart of module helps.
[image]

And related question about `Fade`. Is it possible to disable it for `Power on` and leave only for `Power off`?

smadds, 2017-11-06:
Is it OK with the 5.9.0 release? I am using that on a H801 and it's working fine (once I remapped and enabled the 2 pins for CW & WW!)

dimitrystd, 2017-11-07:
Unfortunately i see the same error. 
* Cloned `v5.9.0` tag
* Added my custom configuration [user_config_override.h](https://github.com/dimitrystd/Sonoff-Tasmota/blob/h801_bedroom/sonoff/user_config_override.h) to `/sonoff` folder
* Compiled and flashed my H801
* Run `SetOption15 1` and waited for restart
* Run `WakeupDuration 1`

See exact the same error in console and also attached output from serial [serial-output.txt](https://github.com/arendst/Sonoff-Tasmota/files/1448451/serial-output.txt)
```
21:16:18 MQT: tele/h801_bedroom/INFO1 = {"Module":"H801", "Version":"5.9.0", "FallbackTopic":"h801_bedroom_4210BF", "GroupTopic":"sonoffs"}
21:16:18 MQT: tele/h801_bedroom/INFO2 = {"WebServerMode":"Admin", "Hostname":"h801_bedroom-4287", "IPAddress":"192.168.1.55"}
21:16:18 MQT: tele/h801_bedroom/INFO3 = {"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x4000bf64 epc2:0x00000000 epc3:0x40204324 excvaddr:0x00000001 depc:0x00000000"}
```

dimitrystd, 2017-11-07:
>once I remapped and enabled the 2 pins for CW & WW!
@smadds can you tell the commands for the console, not sure what are you talking about.

arendst, 2017-11-07:
Empty console verified! Will investigate but from experience it means the logging is corrupt probably due to buffer overflow. My fault. Will check one of these days. (My arilux fails too as soon as I enter command wakeupduration only).

arendst, 2017-11-07:
Quick fix

Replace line 902 in file xdrv_snfled.ino from
```
    snprintf_P(mqtt_data, sizeof(mqtt_data), S_JSON_COMMAND_NVALUE, Settings.light_wakeup);
```
into
```
    snprintf_P(mqtt_data, sizeof(mqtt_data), S_JSON_COMMAND_NVALUE, command, Settings.light_wakeup);
```

dimitrystd, 2017-11-07:
Tested, looks good

closed 2017-11-07 by dimitrystd