arendst/Tasmota #1135: PowerOnState in only ONE special configuration not working? (GPIO5, Relay1, Wemos D1)

stefanbode, 2017-11-06:
Hi gents, can some one please doublecheck the following issue:
Board: Wemos D1 mini
D1 (GPIO5) connected to Relay1

In this case after TOGGLE the device ON and reboot the device is OFF (version 5.8.0q). The PowerOnState 3 does not work in this configuration. As soon as I connect any other GPIO it is working and the state of Relay1 is preserved. Also adding additional relays it worked for Relay2++, but not relay1. Also, reset configuration with "RESET 2" to ensure no other configuration issues.

stefanbode, 2017-11-06:
The problem seems not be in the code. It seems to be dependent on the physical wire-ing. I changed it to D2 and now the "problem" is on D2. After the initial restart, the GPIO is HIGH but reported LOW. Then after an additional reboot, the GPIO is LOW and reported LOW. I do not see where the physical wires can influence the program. If I remove my WEMOS from the board the PowerOnState is working.

davidelang, 2017-11-06:
either the page on troubleshooting or on adding things to your sonoff in the 
wiki.

stefanbode, 2017-11-10:
I need to reopen the issue @davidelang and @arendst. I can verify, that with firmware version 5.5 and similar it was not a problem if one GPIO was more or less connected to GND. With the current firmware the Relay change OFF after reboot (ONLY in interface, is PHYSICAL ON). Then after another reboot is OFF physically and in the interface, too. I assume there is a change that causes this behavior.  If there is a 3k resistor it seems to work.

stefanbode, 2017-11-10:
Here is the log file: you see it goes OFF after reboot, but still reports DS1 and DS2. This ONLY works if the Transistor gets an ON signal. If toggle manually this works without problems:

```
17:07:44 HTP: Main Menu
17:07:44 HTP: Main Menu
17:07:44 HTP: Main Menu
17:07:46 MQT: hm/status/ESP_19F94F/RESULT = {"POWER":"ON"}
17:07:46 MQT: hm/status/ESP_19F94F/POWER = ON (retained)
17:07:46 MQT: hm/status/ESP_19F94F/RESULT = {"POWER":"OFF"}
17:07:46 MQT: hm/status/ESP_19F94F/POWER = OFF (retained)
17:07:51 MQT: hm/status/ESP_19F94F/RESULT = {"POWER":"ON"}
17:07:51 MQT: hm/status/ESP_19F94F/POWER = ON (retained)
17:07:52 CFG: Saved to flash at 3F7, Count 67, Bytes 1648
17:07:54 HTP: Console
17:08:03 SHT: Sensor did not ACK command
17:08:03 WIF: Checking connection...
17:08:03 WIF: Connected
17:08:04 MQT: hm/status/ESP_19F94F/STATE = {"Time":"2017-11-10T17:08:04", "Uptime":0, "POWER":"ON", "Wifi":{"AP":1, "SSId":"BodeNet_Keller", "RSSI":60, "APMac":"38:10:D5:14:E6:70"}}
17:08:04 MQT: hm/status/ESP_19F94F/SENSOR = {"Time":"2017-11-10T17:08:04", "Analog0":2, "DS18x20":{"DS1":{"Type":"DS18B20", "Address":"28405D28000080F2", "Temperature":9.6}, "DS2":{"Type":"DS18B20", "Address":"287B9514060000FC", "Temperature":14.2}}, "TempUnit":"C"}
17:08:20 CMD: restart 1
17:08:20 RSL: Received Topic /restart, Data Size 1, Data 1
17:08:20 RSL: Group 0, Index 1, Command RESTART, Data 1
17:08:20 MQT: hm/status/ESP_19F94F/RESULT = {"Restart":"Restarting"}
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on ESP_19F94F-6479 with IP address 192.168.178.63
00:00:04 MQT: Verify TLS fingerprint...
00:00:04 MQT: Verified
00:00:06 APP: (UTC) Fri Nov 10 16:08:29 2017
00:00:06 APP: (DST) Sun Mar 26 02:00:00 2017
00:00:06 APP: (STD) Sun Oct 29 03:00:00 2017
17:08:30 MQT: Attempting connection...
17:08:30 MQT: Connected
17:08:30 MQT: hm/status/ESP_19F94F/LWT = Online (retained)
17:08:30 MQT: hm/setting/ESP_19F94F/POWER = 
17:08:30 MQT: Subscribe to hm/setting/ESP_19F94F/#
17:08:30 MQT: Subscribe to hm/setting/sonoffs/#
17:08:30 MQT: Subscribe to cmnd/DVES_19F94F/#
17:08:30 RSL: Received Topic hm/setting/ESP_19F94F/deepsleep, Data Size 1, Data 0
17:08:30 RSL: Group 0, Index 1, Command DEEPSLEEP, Data 0
17:08:30 MQT: hm/status/ESP_19F94F/RESULT = {"Command":"Unknown"}
17:08:30 MQT: hm/status/ESP_19F94F/INFO1 = {"Module":"WeMos D1 mini", "Version":"5.9.1a", "FallbackTopic":"DVES_19F94F", "GroupTopic":"sonoffs"}
17:08:30 MQT: hm/status/ESP_19F94F/INFO2 = {"WebServerMode":"Admin", "Hostname":"ESP_19F94F-6479", "IPAddress":"192.168.178.63"}
17:08:30 MQT: hm/status/ESP_19F94F/INFO3 = {"RestartReason":"Software/System restart"}
17:08:33 MQT: hm/status/ESP_19F94F/RESULT = {"POWER":"OFF"}
17:08:33 MQT: hm/status/ESP_19F94F/POWER = OFF (retained)
17:08:40 SHT: Sensor did not ACK command
17:08:41 MQT: hm/status/ESP_19F94F/STATE = {"Time":"2017-11-10T17:08:41", "Uptime":0, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"BodeNet_Keller", "RSSI":64, "APMac":"38:10:D5:14:E6:70"}}
17:08:41 MQT: hm/status/ESP_19F94F/SENSOR = {"Time":"2017-11-10T17:08:41", "Analog0":2, "DS18x20":{"DS1":{"Type":"DS18B20", "Address":"28405D28000080F2", "Temperature":9.6}, "DS2":{"Type":"DS18B20", "Address":"287B9514060000FC", "Temperature":14.3}}, "TempUnit":"C"}
```

stefanbode, 2017-11-10:
Here some additional configuration info, like the SaveState
```

17:31:16 RSL: STATUS = {"Status":{"Module":18, "FriendlyName":"Fensterbank", "Topic":"ESP_19F94F", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":1, "PowerRetain":1}}
17:31:16 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":0, "Sleep":0, "BootCount":33, "SaveCount":73, "SaveAddress":"3F9000"}}
17:31:16 RSL: STATUS2 = {"StatusFWR":{"Version":"5.9.1a", "BuildDateTime":"2017-11-10T16:59:40", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
17:31:16 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2, "WebLog":5, "SysLog":0, "LogHost":"XXXXXXXXXXX", "LogPort":514, "SSId1":"XXXXXXXXXXX", "SSId2":"XXXXXXXXXXX", "TelePeriod":30, "SetOption":"558000E1"}}
17:31:16 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":502, "Free":2568, "Heap":21, "ProgramFlashSize":4096, "FlashSize":4096, "FlashMode":3}}
17:31:16 RSL: STATUS5 = {"StatusNET":{"Hostname":"ESP_19F94F-6479", "IPAddress":"192.168.178.63", "Gateway":"192.168.178.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.178.1", "Mac":"XXXXXXXXXXX", "Webserver":2, "WifiConfig":2}}
17:31:16 RSL: STATUS7 = {"StatusTIM":{"UTC":"Fri Nov 10 16:31:16 2017", "Local":"Fri Nov 10 17:31:16 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":99}}
17:31:16 RSL: STATUS10 = {"StatusSNS":{"Time":"2017-11-10T17:31:16", "Analog0":2}}
17:31:16 RSL: STATUS11 = {"StatusSTS":{"Time":"2017-11-10T17:31:16", "Uptime":0, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"XXXXXXXXXXX", "RSSI":64, "APMac":"XXXXXXXXXXX"}}}
17:31:17 SHT: Sensor did not ACK command
```

davidelang, 2017-11-10:
I can see a problem connecting an output pin directly to ground, you will draw 
the max power possible from the chip (possibly damaging it), if you are using it 
as an input, there should be no problems

why are you connecting this GPIO to ground and how is it configured 
(LED/RELAY/SWITCH/?)

stefanbode, 2017-11-10:
it is configured as RELAY1 and is connected to a transistor BC336-40, but I missed the resistor in the base line. Therefore it is "nearly" direct connected to ground. Anyhow it works. And as mentioned in 5.5 the state was also correct

stefanbode, 2017-11-10:
As you can see in the weblog attached the software is reporting an OFF, but physically the GPIO is HIGH and ON. If you toggle the Relay everything is fine. HIGH and LOW correspond with the software status. There is no issue, if it is LOW (OFF) and you restart. Then is is still LOW and physically also the GPIO is LOW.

boaschti, 2017-11-18:
I have a similar Problem with my GPIO16. My schematic is here.
https://github.com/boaschti/MQTT_Mini_WLan_Relais/blob/master/Board/typical_pinning.JPG
If i restart the ESP then the relay is in toggled state. 
It seems that tasmota trigger an toggle if SwitchMode<x> = 0 at restart. 
If i use a other Port it works, so the connected Hardware seems to make a difference in behavior.

stefanbode, 2017-11-28:
The issue is created by two changes required by #526 and #909 . I have the same wrong situation with a BC337 transistor. There is no resistor in the base. Therefore if the relay is set ON, in sonoff.ino line 2732ff the digital read switches the status to the unwanted situation (OFF). after a reboot the status is off:

@arendst: If the behavior is intended, at least you should take care that reported and real status is not out of sync. You can reproduce by connecting D1 to GND, Set Relay1 ON. Reboot: Boom!
```
  // Issue #526 and #909
  for (byte i = 0; i < devices_present; i++) {
    if ((i < MAX_RELAYS) && (pin[GPIO_REL1 +i] < 99)) {
      bitWrite(power, i, digitalRead(pin[GPIO_REL1 +i]) ^ bitRead(rel_inverted, i));
    }
    if ((i < MAX_PULSETIMERS) && bitRead(power, i)) {
      pulse_timer[i] = Settings.pulse_timer[i];
    }
  }
```
The problem does not occur, if a resistor is present. In this case the digitalRead reports high and everything is fine. To be honest I do not really understand the two addresses issues. In my opinion, the Settings.power has always the right status, therefore not need to read the PIN and change something. Especially, if IN FACT during the first start the real status and the reported one is NOT in sync. Here my log sile with many additional logs to get an idea where the status is changed accidentally:

You see relay is on: 00:00:00 SetDevicePower: GPIO 17, PIN 5, STATE 1, rPower 1
Then the digital read out-sync it: 00:00:00 setup: pulsetimer setup i:0 pin read: 0
```
00:00:00 SettingsLoad: settingpower after 1
00:00:00 SettingsLoad: settingpower after 1
00:00:00 SettingsLoad: settingpower after 1
00:00:00 SettingsLoad: settingpower after 1
00:00:00 SettingsLoad: settingpower after 1
00:00:00 CFG: Loaded from flash at 3F6, Count 134
00:00:00 SettingsLoad: settingpower 1
00:00:00 APP: Boot Count 116
00:00:00 setup: resetInfo.reason 5
00:00:00 setup: power something strange saved: 1
00:00:00 SetDevicePower: rpower 1
00:00:00 SetDevicePower: GPIO 17, PIN 5, STATE 1, rPower 1
00:00:00 SetDevicePower: after USE_PCF8574, PIN 0, STATE 1, rPower 1
00:00:00 setup: pulsetimer setup i:0 pin read: 0
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_1935C2, GroupTopic sonoffs) Version 5.9.1f
00:00:00 setup: finally settingpower 1, power: 0
00:00:00 case (STATES/10)*2: power 0, settings.power 1
00:00:00 SettingsSave: start settingpower 0
00:00:00 CFG: Saved to flash at 3F5, Count 135, Bytes 1664
00:00:00 SettingsSave: settingpower 0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 BodeNet_Keller in mode 11N as sonoff-5570...
00:00:01 case (STATES/10)*2: power 0, settings.power 0
00:00:01 SettingsSave: start settingpower 0
00:00:01 SettingsSave: settingpower 0
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 case (STATES/10)*2: power 0, settings.power 0
00:00:02 SettingsSave: start settingpower 0
00:00:02 SettingsSave: settingpower 0
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...

```

boaschti, 2017-11-29:
In my case there is a resistor between port and mosfet and a second resistor parallel to S and G to have a defined level when port is floating. See in schematic https://github.com/boaschti/MQTT_Mini_WLan_Relais/blob/master/Board/typical_pinning.JPG

stefanbode, 2017-11-29:
I checked your layout and it looks fine for me. Also you should not be affected by the digitalRead issue. Maybe you just add s
ome logging before and after the digitalRead and get an idea, if this changes your relay state. I did not found any additional strange code that could change the behavior.

closed 2018-05-07 by stale[bot]