arendst/Tasmota #1136: Add 2nd relay to Sonoff Basic

BastianJaegerIntel, 2017-11-06:
I would like to add an additional relay to a separate GPIO of the Sonoff basic. 
So far I got the idea that it is possible, but I have not found any instructions how this can be realized. 
- What GPIO can I use?
- What changes in the code are required?
(I am quiet familiar with c/c++, but obviously not with this code.)

Relay would be like this:
http://driscocity.com/wp-content/uploads/2014/05/connectrelay1.jpg

Thx in advance!

davidelang, 2017-11-06:
look at the wiki under the section 'adding things to the sonoff" or something 
like that (down near the bottom of the list)

no code changes are needed, just wire it up and configure the software to know 
what you wired

BastianJaegerIntel, 2017-11-06:
Not sure what you mean?
https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs called "Wiring additional things to a module" in the menu?
How to "configure the software"?

davidelang, 2017-11-06:
On Mon, 6 Nov 2017, BastianJaegerIntel wrote:

> Not sure what you mean?
> https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs called "Wiring additional things to a module" in the menu?

yes

> How to "configure the software"?

on the web interface, it's under config -> module where you define what is 
hooked to each GPIO

BastianJaegerIntel, 2017-11-06:
On the web interface? Ok. great. I have not seen that. 
It will be a start. 
I would like to enable it on default. 

Thanks for the quick reply!

smadds, 2017-11-06:
Any configuration setting you make are stored in non-volatile memory, @BastianJaegerIntel, so the additional relay will be enabled by default once set. You'll need a transistor to drive the relay (and a diode across the coil to protect the transistor), as the ESP won't provide enough current to drive it directly.

jetema, 2017-11-06:
Read this #738 to add succesfully an additional relay to a separate GPIO of the Sonoff basic

BastianJaegerIntel, 2017-11-07:
Thanks for all the replys! Thats a great start and the SW is 100% great! Now a better (2-way) relay solves my other issues. :)

@smadds Sure. I was not precise enough. I meant by "default" that after flashing the device e.g. GPIO04 is already set as Relay2.

davidelang, 2017-11-07:
On Tue, 7 Nov 2017, BastianJaegerIntel wrote:
>
> @smadds Sure. I was not precise enough. I meant by "default" that after flashing the device e.g. GPIO04 is already set as Relay2.

you can compile your own version and change the definition of the module type in 
sonoff.ino to be whatever you want.

David Lang

closed 2018-05-07 by stale[bot]

gishanc, 2020-11-10:
Problem is if Sonoff's power supply is enough to power 2 relays (long term) as it is designed only for 1 relay? What do u think about that?

ascillato, 2020-11-10:
The power regulator of a sonoff basic is very weak. It is designed to be at the edge of the energy it needs. So, as explained in other issues and as the experience of other users, it won't work or it will fail eventually. A sonoff basic is designed only to act as a sonoff basic. If you need 2 relays, use a sonoff dual, a shelly 2.5 or a nodemcu with a 2 relays board.

brasofilo, 2022-12-03:
Years later, thanks to all involved in this discussion.
I managed to add a temperature sensor but when I tried to add an extra relay, things fell apart.
`@ascillato` answer closes the deal for me.

twanjaarsveld, 2024-08-07:
I need 2 extra relays to control a ceiling fan
Should this be possible ?

sfromis, 2024-08-07:
Not on a Sonoff Basic, it does not have the power to engage multiple relays, but you can get similar devices (different shape) with two relays, or boards with multiple.