arendst/Tasmota #1138: Arduino compiling error

hitter99, 2017-11-06:
I'm new to the Sonoff-Tasmota group and I'm trying to setup the Arduino enviroment like described in the Wiki here. I picked the "non-admin" install and followed each step. 
I can't figure out what I'm doing wrong. All settings seem to be correct (like the IDE-parameters mentioned), but when I try to "Verify/Compile" I get the following error message which looks It looks like a missing library :

Arduino: 1.8.5 (Windows 7), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DOUT, 115200, 1M (no SPIFFS), ck, Disabled, None"

N:\APPS\ArduinoSonoff\arduino-builder -dump-prefs -logger=machine -hardware N:\APPS\ArduinoSonoff\hardware -hardware N:\APPS\ArduinoSonoff\portable\packages -tools N:\APPS\ArduinoSonoff\tools-builder -tools N:\APPS\ArduinoSonoff\hardware\tools\avr -tools N:\APPS\ArduinoSonoff\portable\packages -built-in-libraries N:\APPS\ArduinoSonoff\libraries -libraries N:\APPS\ArduinoSonoff\portable\sketchbook\libraries -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dout,UploadSpeed=115200,FlashSize=1M0,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10805 -build-path C:\Users\nlv08193\AppData\Local\Temp\arduino_build_228035 -warnings=none -build-cache C:\Users\nlv08193\AppData\Local\Temp\arduino_cache_427778 -prefs=build.warn_data_percentage=75 -verbose N:\APPS\ArduinoSonoff\portable\sketchbook\sonoff\sonoff.ino
N:\APPS\ArduinoSonoff\arduino-builder -compile -logger=machine -hardware N:\APPS\ArduinoSonoff\hardware -hardware N:\APPS\ArduinoSonoff\portable\packages -tools N:\APPS\ArduinoSonoff\tools-builder -tools N:\APPS\ArduinoSonoff\hardware\tools\avr -tools N:\APPS\ArduinoSonoff\portable\packages -built-in-libraries N:\APPS\ArduinoSonoff\libraries -libraries N:\APPS\ArduinoSonoff\portable\sketchbook\libraries -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dout,UploadSpeed=115200,FlashSize=1M0,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10805 -build-path C:\Users\nlv08193\AppData\Local\Temp\arduino_build_228035 -warnings=none -build-cache C:\Users\nlv08193\AppData\Local\Temp\arduino_cache_427778 -prefs=build.warn_data_percentage=75 -verbose N:\APPS\ArduinoSonoff\portable\sketchbook\sonoff\sonoff.ino
Using board 'generic' from platform in folder: N:\APPS\ArduinoSonoff\portable\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: N:\APPS\ArduinoSonoff\portable\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
recipe.preproc.macros pattern is missing

Error compiling for board Generic ESP8266 Module.

jetema, 2017-11-06:
If it is hard to find where is the fault, you also can try to Flash firmware binary directly ( sonoff.bin file )  from your PC to the sonoff with no compiling at all.

marvinwaterman, 2017-11-18:
The fault is that you have the newer version of IRRemoteESP8266 library installed.  You will need to change the lines inside of "xdrv_irremote.ino" to read as below. (Note: we have to include IRsend.h and IRrecv.h according to this page https://github.com/markszabo/IRremoteESP8266/wiki/Upgrading-to-v2.0)

#ifndef USE_IR_HVAC
#include <IRremoteESP8266.h>
#include <IRsend.h>  // Needed if you want to send IR commands.
#include <IRrecv.h>  // Needed if you want to receive IR commands.
#else
#include <IRMitsubishiAC.h>

skatun, 2017-11-23:
I get the same error:

recipe.preproc.macros pattern is missing.

Seems to be folder structure errors

pdeutsch, 2017-12-05:
I just setup my environment for the first time too and am getting the same error.
```
recipe.preproc.macros pattern is missing

Error compiling for board Generic ESP8266 Module.
```

My setup is a Lubuntu desktop with Arduino IDE 1.8.0.

itoutline, 2017-12-25:
If you perform procedure step by step from https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE then "esp8266 by ESP8266 Community" package won't be fully installed.  Just remove portable\packages\esp8266\ folder. Install the package esp8266 from Boards Manager and then copy eagle.flash.1m0.ld and boards.txt to proper folders. In my case it was the source for "recipe.preproc.macros pattern is missing"

closed 2018-05-08 by stale[bot]