arendst/Tasmota #1139: Sonoff S20 Smart Socket won't work, flashes OK

MatsKarlsson, 2017-11-06:
I Successfully flash the S20 Smart Socket. ...100%
But when it won't log anything in arduino serial log after.

Putting it in a socket doesn't send any mqtt messages. pressing button does not switch it on or off.

It's a ESP8266 chip on it, board rev is: S20_EU_V1.3, with a date of 2017-2-16

Have flashed a ton of various Sonoff devices but can't seem to find whats the problem with this. (have tried 2 different S20 Smart sockets, same with both)

MatsKarlsson, 2017-11-06:
Sorry, my bad :-( 

Flashed with DIO instead of DOUT, reflashed with correct flash mode, now it works like a charm.

closed 2017-11-06 by MatsKarlsson

RoeBus, 2017-11-10:
Hello Mats,
Hopefully this works (after closing topic).. I have been digging around the internet for days and came across your issue. I have the same issue, two units wont be flashed, no matter what. Can you share the command that worked for you? What image did you flash?

Thanks,
Roel

strasharo, 2017-11-10:
Did you try with https://github.com/mirko/SonOTA ? I already flashed four S20s without any issues.

MatsKarlsson, 2017-11-10:
Hello @RoeBus ,
Now it works for me, I just downloaded the latest code from this repository.

In Arduino IDE you can set "flash mode", I think it's `DIO` by default, change that so it is: `DOUT`

When using wrong flash mode I could flash the devices, but they didn't work after I flashed them.

it even says so on the front page of Tasmota:

----------

ATTENTION All versions

**Only Flash Mode DOUT is supported**. Do not use Flash Mode DIO / QIO / QOUT as it might seem to brick your device.

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips) for background information.

davidelang, 2017-11-10:
On Fri, 10 Nov 2017, MatsKarlsson wrote:

> **Only Flash Mode DOUT is supported**. Do not use Flash Mode DIO / QIO / QOUT as it might seem to brick your device.

it's a soft brick, you can fix it with a flash over serial. nothing you can do 
in software will prevent that from working.

David Lang