arendst/Tasmota #1142: Sonoff Basic Reed Garage Sensor

jannnfe, 2017-11-06:
Hello,

I'm new to this area so please forgive me beginner questions. I have already flashed My Sonoff Basic and have already set it up so successfully that I can open my garage door. I set Pulsetime so that the Sonoff only switches for 1 sec, a Ralais activates that gives the garage a close contact. It is working perfectly with Homebridge and tasmota plugin.

Now I want to add a reed sensor that I always know when the garage is up and down. I would prefer if in the Apple Home App, the button is displayed turn on as long as the garage is open. As soon as the Reed sensor gets a contact, the button in the app turns off.

I have already connected the Reed Sensor to GPIO14 and when I set this in the Tasmota firmware via the web interface as a switch, he switches the Sonoff.

I hope someone understands what I mean and can tell me exactly what I need to do to make it work. Please with exact MQTT commands or other.

Many thanks in advance.

ionciubotaru, 2017-11-07:
read about switchtopic and switchmode - I have a garage with a reed sensor and works like a charm

jannnfe, 2017-11-07:
Yes, I already thought it had something to do with what you said. However, I do not know exactly where to enter the commands. In the console in the web interface?
Likewise, I'm not quite clear the structure of the commands. Could maybe give you an example?

ionciubotaru, 2017-11-07:
You cand send commands via serial interface, web interfacte if activated or mqtt. Try:

`cmnd/garage/switchtopic garagedoor`
`cmnd/garage/switchmode 1`

Then check the messages you receive via mqtt when the reed relay is close or open

oetelaar24, 2017-11-07:
Hi, I have the exact same problem as @jannnfe 
Using a Sonoff Basic, to test with a Reed switch.. Configured the switch als a switch1 on GPIO14. connected on pin 4 and 5..
It doesn't seem to matter what I configure, have a switchtopic set or not, on 0,1 or 2. Setting the switchmode to 0,1 or anything else. Whatever I try, the reedswitch allways operates the Relais in the Sonoff. Exactly what I don't want...
Using the newest master firmware 5.9.1... I must be doing something wrong.. but what?

davidelang, 2017-11-08:
you set switchtopic to the topic that you  want to have the message stnt as (it 
must be different than the topic for the device)

k

oetelaar24, 2017-11-08:
Thanks for your comment David, however, that doesn't work. It doesn't matter what I set as switchtopic, the relais always reacts to it...

davidelang, 2017-11-08:
show us your config

oetelaar24, 2017-11-08:
00:03:11 CMD: gpio
00:03:11 RSL: stat/sonoff/RESULT = {"GPIO1":"0 (None)", "GPIO3":"0 (None)", "GPIO4":"0 (None)", "GPIO14":"9 (Switch1)"}
00:03:30 CMD: switchtopic
00:03:30 RSL: stat/sonoff/RESULT = {"SwitchTopic":"garagedeur"}
00:03:33 CMD: switchmode
00:03:33 RSL: stat/sonoff/RESULT = {"SwitchMode1":1}

Then when using the Reed-switch on GPIO14, i see:
00:04:54 RSL: stat/sonoff/RESULT = {"POWER":"ON"}
00:04:54 RSL: stat/sonoff/POWER = ON
00:04:55 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
00:04:55 RSL: stat/sonoff/POWER = OFF

oetelaar24, 2017-11-13:
Anyone any idea?

jannnfe, 2017-11-21:
I have now attached a sensor to my garage, which recognizes the status "closed". This also works great with @ionciubotaru solution.

Now I want to extend the whole thing a bit. I would like to use the plugin https://github.com/iomax/homebridge-mqttgaragedoor to control Apple's home app.

I have the following configuration:
"topics": {
        "statusSet": "cmnd/Garagentor/POWER",
        "openGet": "cmnd/Garagentor_Sensor/POWER1",
        "closedGet": "cmnd/Garagentor_Sensor/POWER1",
        "openValue": "ON",
        "closedValue": "OFF"
      }

but with these settings, it does not work perfectly. I think it's because I've set Pulstime in the Tasmota firmware and every time I open the Sonoff first POWER on and then directly sends POWER off again.

It would have to go, that you drive up the garage with Pulstime, the sonoff reports POWER on and if I press again it sends POWER off independently of the Pulstime. Can I possibly define other onValue and offValue?

oetelaar24, 2017-12-11:
Ok, so, after af few weeks of other things, I decided, heck, maybe I just need to configure an MQTT server... And yes.. That did the trick. Basically, the sonoff needs to be able to connect to an MQTT server, otherwise it doesn't work at all. Now the reed switch works brilliant :-D

closed 2018-01-16 by jannnfe

Alfaspider73, 2018-01-25:
Hello !
Can someone help me with the config.json ? Get this error displayed in homebridge.

TypeError: Cannot read property 'length' of undefined
    at Object.validateTopics (/usr/local/lib/node_modules/homebridge-mqttgaragedoor/node_modules/mqtt/lib/validations.js:42:30)
    at MqttClient.subscribe (/usr/local/lib/node_modules/homebridge-mqttgaragedoor/node_modules/mqtt/lib/client.js:417:30)
    at new MqttGarageAccessory (/usr/local/lib/node_modules/homebridge-mqttgaragedoor/index.js:86:17)
    at Server._loadAccessories (/usr/local/lib/node_modules/homebridge/lib/server.js:275:29)
    at Server.run (/usr/local/lib/node_modules/homebridge/lib/server.js:86:38)
    at module.exports (/usr/local/lib/node_modules/homebridge/lib/cli.js:40:10)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge/bin/homebridge:17:22)
    at Module._compile (module.js:635:30)
    at Object.Module._extensions..js (module.js:646:10)
    at Module.load (module.js:554:32)
I'm sure I have another error in Config.json. That's my config.json
{
   "bridge":{
      "name":"Homebridge",
      "username":"CC:22:3D:E3:CE:30",
      "port":51826,
      "pin":"031-45-154"
   },
   "platforms":[
      {
         "platform":"TplinkSmarthome",
         "name":"TplinkSmarthome"
      },
      {
         "platform":"Hue",
         "users":{
            "001788FFFEA29534":"ARJW6OJDrtdGi65NPNSplJAK5LkfuiWHvrBgCtbW"
         },
         "lights":true
      }
   ],
   "accessories":[
      {
         "accessory":"Mqttgarage",
         "name":"Garagentor",
         "url":"mqtt://192.168.178.00",
         "username":"",
         "password":"",
         "caption":"Garageport",
         "topics":{
            "statusGet":"stat/sonoff/RESULT",
            "statusSet":"cmnd/Garagentor/POWER",
            "openGet":"cmnd/Garagentor_Sensor/POWER1",
            "closedGet":"cmnd/Garagentor_Sensor/POWER1",
            "openValue":"ON",
            "closedValue":"OFF"
         }
      }
   ]
}

Hope someone can help me. 
I am still beginner

jannnfe, 2018-01-25:
Hello @Alfaspider73,
Your config.json seems to be valid.
I have learned a lot by now and just use the homebidge-mqtt plugin, which I use to control my garage door via Node-Red. I would recommend you the same system. I had planned to share my garage system in the homebridge-mqtt wiki if anyone is interested.
Already one thing ahead. You need two reed sensors in your garage to make the system reliable enough.

Alfaspider73, 2018-01-25:
Thanks for the reply 
I must look me how this works 
With Node-red I don't know my

Alfaspider73, 2018-02-05:
Hello @jannnfe !
I have great interest in your implementation. Can you help me and explain how you did that ? Have a Sonoff basic with Tasmota and 2 Reed contacts. Node-red I have already installed but'm so still not so good. Could you help me? Excuse me for my bad English

jannnfe, 2018-02-05:
Hello,
look here: https://github.com/jannnfe/homebridge-ESP8266-pojects
In the next few days I will continue to expand my directory and then you should also find an example of the garage opener!

Alfaspider73, 2018-02-06:
Super, I'm already very excited

jannnfe, 2018-02-06:
So I've already finished the biggest part. It's a lot of work, more work than I thought. I'm not quite ready but I'll add the rest later:
https://github.com/jannnfe/homebridge-ESP8266-pojects/wiki/Itead-Sonoff-Garagedoor-Opener

Alfaspider73, 2018-02-07:
Great work 👍👍👍