arendst/Tasmota #1143: Sonoff Basic hardware button press

xenio, 2017-11-07:
Is it possible to know if the button on the Sonoff basic is manually pressed?
I get always the same message:
stat/sonoffbasic1/RESULT
{"POWER":"ON"}
for the hardware button, for the toggle button on the web page and also if I send a mqtt message to power on.
I need to know if the hardware button is pressed because I want to override the bigtimer on node-red flow.
Thanks.

davidelang, 2017-11-07:
when you press and release the button it will switch the relay, or if you 
set buttontopic it will send a message

by default, it will do different things if you press it multiple times quickly 
or hold it for more than 4 seconds.

xenio, 2017-11-07:
Thanks for your help, as you can see I am new to mqtt and sonoff, I did some test and this is what I get:
I sent "/cmnd/sonoffbasic1/ButtonTopic switchsonoffbasic1" to the sonoff
and I get: "stat/sonoffbasic1/RESULT" > {"ButtonTopic":"switchsonoffbasic1"}

Now when I press the hardware button, the relay and the led does nothing and I get:
"cmnd/switchsonoffbasic1/POWER" > TOGGLE
is this "TOGGLE" that I have to process with Node-RED?

closed 2018-05-07 by stale[bot]