arendst/Tasmota #1146: Command "Dimmer 100" breaks H801 device

dimitrystd, 2017-11-07:
*Maybe it's related to #1133*
Spent a couple days to understand why H801 hangs from time to time.

* Use `v5.9.0a`
* Added my custom configuration [user_config_override.h](https://github.com/dimitrystd/Sonoff-Tasmota/blob/h801_bedroom/sonoff/user_config_override.h) to `/sonoff` folder
* Compiled and flashed my H801
* Run `SetOption15 1` in console and waited for restart
* Send via mqtt `cmnd/h801_bedroom/Dimmer 5`. Everything looks good
* Send again 100 via mqtt `cmnd/h801_bedroom/Dimmer 100`

Device starts to restart in infinite loop. It doesn't connect to wifi so i canno open console. In serial i see only
```

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
....
```
Turn off then on doesn't help. I have to flash device with blank image and then flash with firmware.

dimitrystd, 2017-11-07:
If run command in console
```
10:12:49 CMD: Dimmer 100
10:12:49 MQT: stat/h801_bedroom/RESULT = {"POWER":"ON", "Dimmer":100, "Color":"FFFFFF"}
```
Then it just restart once
```
10:13:06 MQT: tele/h801_bedroom/INFO3 = {"RestartReason":"Hardware Watchdog"}
```
So you should use mqtt for reproducing.

dimitrystd, 2017-11-07:
Hmm, repeated test with console, namely
```
speed 3
fade on
dimmer 100
```
and now it's died again.

arendst, 2017-11-07:
What is your PWMfrequency and PWMRange?

dimitrystd, 2017-11-07:
```
10:37:02 CMD: PWMfrequency
10:37:02 MQT: stat/h801_bedroom/RESULT = {"PWMFrequency":880}
10:37:10 CMD: PWMRange
10:37:10 MQT: stat/h801_bedroom/RESULT = {"PWMRange":1023}
```
Just tested on `5.9.0b` and bug is still here
```
10:36:09 MQT: tele/h801_bedroom/INFO1 = {"Module":"H801", "Version":"5.9.0b", "FallbackTopic":"h801_bedroom_4210BF", "GroupTopic":"sonoffs"}
```

arendst, 2017-11-07:
How about dimmer 99

dimitrystd, 2017-11-07:
Used console and mqtt, in both cases it works fine.

arendst, 2017-11-07:
For the Sonoff Led I clip dimmer at 99 as a value of 100 made the leds blink too much when the PWM pulses are probably non existent. I guess the same is true for the H801 but it shows as a Watchdog timeout.

I think I will clip dimmer at 99 for all other too. You won't notice any way... ;-)

arendst, 2017-11-07:
You might want to try the set pwmfrequency lower and see what happens with dimmer 100.

I have a Arilux which blinks fast at pwmfrequency 2000 and dimmer 100 but is steady at dimmer 99

dimitrystd, 2017-11-07:
```
12:25:37 CMD: PWMfrequency 600
12:25:37 MQT: stat/h801_bedroom/RESULT = {"PWMFrequency":600}
12:25:49 CMD: Dimmer 99
12:25:49 MQT: stat/h801_bedroom/RESULT = {"POWER":"ON", "Dimmer":99, "Color":"FCFCFC"}
12:25:55 CMD: Dimmer 100
12:25:55 MQT: stat/h801_bedroom/RESULT = {"POWER":"ON", "Dimmer":100, "Color":"FFFFFF"}
```
Again endless reboot.

My main pain point that there is no way to reset device. It's physically mounted under the bed. I have to deinstall it and to flash it two times (with blank and actual firmware) then do a couple additional adjustments that cannot be configured via code.

So, for now, i will change openhab configuration to never send 100. But would be nice to have the option to reset configuration without device deinstallation.

arendst, 2017-11-07:
During the reboot you might be able to enter command power off inthe web console. Then you have all the time to change the dimmer value below 100.

dimitrystd, 2017-11-07:
It doesn't connect to wifi, so i cannot use console or mqtt. I'm not sure how to send command via serial port, but it also means: deinstall device; unscrew case; connect laptop...

stefan-wiesner, 2018-01-01:
Hi.

This did not solve this issue for me. It was neither the `reset 1` nor `reset 2` nor `esptool.py --port COM5 erase_region 0x0F4000 0x008000` nor `esptool.py --port COM5 erase_flash`... It seemed that I totally messed up my Sonoff.

Fix for me was to download an empty 1MB flash BIN and flash it with the esptool like that:

`python esptool.py --port COM5 write_flash -fs 1MB -fm dout 0x0 blank_1MB.bin`

I put the bank_1MB.bin into the folder where the esptool.py was in. This really wipes the whole megabyte of flash memory.

The page to download the empty 1MB can be found here (link at the very bottom):

http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/

With addressing other areas in the memory it is also possible to wipe 2MB of flash like shown in the article.

@arendst maybe you want to add it to the troubleshooting section - because the erase_flash command did not fix it for me and I was short befor throwing the sonoff into the trash... And thanks for the great Tasmota Software!

Hope it helps ...