arendst/Tasmota #1148: Secure way for internetaccess

freaktech, 2017-11-08:
hi
what would you suggest for remote Access to a sonoff device.

the webinterface Password is not encoded, and i think this is unsecure

do You use remote Access and what do You use, for Opening a door for example.

thanks
Tom

arendst, 2017-11-08:
Accessing the webpages over the internet is insecure and is not recommended.

If you want to access Sonoff off premises I would suggest to use cloud MQTT with TLS. As TLS needs a lot of memory resources the webserver must be disabled to get a reliable device.

Have a look at this https://github.com/arendst/Sonoff-Tasmota/wiki/Securing-your-IoT-from-hacking where security information is explained.

freaktech, 2017-11-08:
sounds realy complicate :-( 
for a newbie

davidelang, 2017-11-08:
that's why it's not recommended :-)

setup your home automation software to be able to be accessed over the Internet, 
secure access to it, and have it manage your devices.

the other thing you can do is to setup a VPN on your router and use that to 
connect to your network to access the devices.

closed 2018-05-07 by stale[bot]