arendst/Tasmota #1150: Split sensor data to MQTT

Siorinex, 2017-11-08:
Add switch for split data from sensor to other MQTT topic:
1 - tele/cooler/SENSOR = {"Time":"2017-11-08T08:50:42", "BME280":{"Temperature":26.7, "Humidity":31.3, "Pressure":1011.0}, "TempUnit":"C"} (CoolerSensor.Sensor)
2 - tele/cooler/SENSOR/BME280/Temperature = 26.7
tele/cooler/SENSOR/BME280/Humidity = 31.3
etc.

arendst, 2017-11-08:
Won't be done as every single line will be a message swamping your network. 

This was the case in earlier generations of this software.

Siorinex, 2017-11-08:
Swamping my network - STATE and LWT message.
Not need for default - only at option: in Majordomo and other MQTT Smart Home System`s - send from MQTT to value - automaticle. But if value have any value included - it`s problem for system!

In other ESP8266 and ESP32 devices I use Wifi-IoT firmware: it`s send data to MQTT spliting and not "swamp network"!

davidelang, 2017-11-08:
pick a home automation tool that can watch for the JSON messages and then break 
them down and spit out as many smaller messages as you want.

As Arendst says, it used to be that tasmota sent separate messages, it was a 
problem so it was changed

state messages should only be sent when something changes, lwt messages should 
only be sent when a device connects or disappears. If those are swamping your 
network, something seems wrong.

closed 2018-05-07 by stale[bot]