arendst/Tasmota #1151: Any chance of compatibility with IoTaWatt - 14 Channel Electricity Monitor

eazuka, 2017-11-08:
Hi @arendst, any chance of Sonoff-Tasmota supporting IoTaWatt?

IoTaWatt is an open-hardware 14 channel WiFi connected electric power monitor based on the ESP8266 IoT platform using MCP3208 12 bit ADCs to sample AC voltage and current with on-board SD card for local storage.

https://shop.openenergymonitor.com/iotawatt-wifi-connected-14-channel-electricity-monitor/
https://blog.openenergymonitor.org/2017/10/iotawatt/

Since it's also based on ESP8266, any chance of it being supported, just like the Sonoff Pow?

arendst, 2017-11-08:
Just looked at it and as Bob Lemaire already has done a great job I see no real reason to duplicate it. I also noticed that timing is a big issue so that makes integration even more difficult.

Nice project though.

eazuka, 2017-11-08:
what do you mean by timing is a big issue and how does it hinder integration?
just asking for me to better understand?

arendst, 2017-11-08:
Dive into the source of the IoTaWatt project and you'll see that every 50/60Hz all analog ports need to be scanned for updates.

eazuka, 2017-11-08:
oh, i see

closed 2017-11-09 by arendst