arendst/Tasmota #1154: Time limit of power ON / OFF states

smhgit, 2017-11-08:
Enable to set time limit for power on / off state. The device will change state after timer expired. (Need this feature because when controlling device such as boiler, I want to make sure that the device will turned off after predefined time even for any reason the off command wasn't received).

jetema, 2017-11-08:
See pulsetime on   https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

PulseTime<x> |   | Show current PulseTime of relay<x> in 0.1 seconds
PulseTime<x> | 0 / off | (Default) Disable use of PulseTime for relay<x>
PulseTime<x> | 1..111 | Set PulseTime for relay<x> with 0.1 seconds increment
PulseTime<x> | 112..64900 | Set PulseTime for relay<x> with 1 seconds increment starting with 12 seconds (113 = 13 seconds etc.)

Hope it helps

smhgit, 2017-11-08:
@jetema thx!

Yes, this help. Does it also support re-triggering? that means, can i start new pulse without toggling?

1. Pulse 60 sec
2. Turn on

After 30 seconds

3. Pulse 60 sec

After 60 seconds sonoff turn off (was on for 90 seconds)

jetema, 2017-11-08:
I'm not sure if you can get  re-triggering, see this 
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#main

smhgit, 2017-11-09:
Seems that re-triggering doesn't supported. Will be nice to add this feature.

stefanbode, 2017-11-16:
How many re-triggers do you need? Endlessly is not possible, but you can send several commands in a chain. That may solve your use case. Otherwise this logic must be enabled outside the device

Frogmore42, 2017-12-23:
The simple cases seem to work fine.  I have a Sonoff that is reporting every 30s and Node-Red then evaluates the conditions and sends an "on" or "off" depending on temperature.  I have set pulse time to 33 second.  Node-Red normally responds fast enough to keep the relay on before the pulse time has expired, but if Node-Red were to stop responding, the Sonoff would still turn off by itself.  

On another one, I have pulse time set to 10s and report period to 30s.  This also works fine, but the Sonoff turns off well before the next iteration is evaluated.

mkh595, 2018-01-11:
_Seems that re-triggering doesn't supported. Will be nice to add this feature._

Via MQTT you can periodically send command POWER ON before PulseTime period is reached and counting starts again . This keeps relay ON.

closed 2018-05-25 by stale[bot]