arendst/Tasmota #1155: Sensor issue with basic

edit21, 2017-11-08:
Hi all, running 5.9.1a
Have sensor connected to GPIO14 though cannot get any output, have #define USE_DS18x20  - compiled as per usual. 
GPIO from the console shows that its setup - its also setup as usual in configure module:
20:04:13 MQT: stat/sonoffsingle3/RESULT = {"GPIO1":"0 (None)", "GPIO3":"0 (None)", "GPIO4":"0 (None)", "GPIO14":"4 (DS18x20)"}

Any idea's?

davidelang, 2017-11-08:
reboot after changing the module definition.

edit21, 2017-11-08:
Thanks David, 
I did do originally and its still not reporting, rebooted again to be sure and still no output

davidelang, 2017-11-08:
turn logging to 'more detail' and see what messages you see during startup. does 
it see the sensor? does it try to initialize the sensor?

ionciubotaru, 2017-11-09:
add a 10K pull up resistor

edit21, 2017-11-09:
00:00:07 MQT: tele/sonoffsingle3/LWT = Online (retained)
00:00:07 MQT: cmnd/sonoffsingle3/POWER = 
00:00:07 MQT: Subscribe to cmnd/sonoffsingle3/#
00:00:07 MQT: Subscribe to cmnd/sonoffs/#
00:00:07 MQT: Subscribe to cmnd/sonoffsingle3/#
00:00:07 MQT: tele/sonoffsingle3/INFO1 = {"Module":"Sonoff Basic", "Version":"5.9.1a", "FallbackTopic":"sonoffsingle3", "GroupTopic":"sonoffs"}
00:00:07 MQT: tele/sonoffsingle3/INFO2 = {"WebServerMode":"Admin", "Hostname":"sonoffsingle3", "IPAddress":"172.16.16.108"}
00:00:07 MQT: tele/sonoffsingle3/INFO3 = {"RestartReason":"Power on"}
00:00:08 MQT: stat/sonoffsingle3/RESULT = {"POWER":"OFF"}
00:00:08 MQT: stat/sonoffsingle3/POWER = OFF
00:00:15 MQT: tele/sonoffsingle3/STATE = {"Time":"1970-01-01T00:00:15", "Uptime":0, "Vcc":3.208, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"Error", "RSSI":80, "APMac":"00:1A:8C:C0:E4:31"}}
00:00:26 WIF: Checking connection...
00:00:26 WIF: Connected

No not even seen

Ion, thanks - where :) ?

ionciubotaru, 2017-11-09:
connect a 10K resistor between gpio14 and 3V3

edit21, 2017-11-10:
10k  did the trick!
Thanks

closed 2017-11-10 by edit21