arendst/Tasmota #1156: RGBW LED strips

jhlweb, 2017-11-08:
Is there any way off adding support for RGBW LED strips. 

If used a Wemos D1 and this gives random colors because it needs a 32bit value for each light and it gets a 24bit value.

sbryk, 2017-11-09:
I think you mean "Neopixel RGBW strips", as the support for dumb RGB strips is already here in 5.9.1 over H801 and Magic Home controller

jhlweb, 2017-11-09:
Yes Neopixel or like RGBW LED strips

arendst, 2017-11-09:
RGBWC pwm controlled ledstrips are supported.
RGB neopixel ledstrips are also supported.
RGBW neopixel ledstrips are not supported.

jhlweb, 2017-11-10:
Hi @arendst I saw it in the code.. 
I upgrade my WS2812 file and used 
`#define USE_WS2812_CTYPE     3                 // WS2812 Color type (0 - RGB, 1 - GRB, 2 - RGBW, 3 - GRBW)`
And did some changes in xdrv_snfled to make it possible to set them by hand. (line 803) but it fails and i'm thinking it's something with light_subtype but i can't figure out what the problem is.  Any idea how to make it work to use LT_ws2812 and also send 4 bytes?

```
        if (USE_WS2812_CTYPE > 1){
          Ws2812SetColor(index, light_entry_color[0], light_entry_color[1], light_entry_color[2], light_entry_color[3]);
        } else {
          Ws2812SetColor(index, light_entry_color[0], light_entry_color[1], light_entry_color[2]);
        }
```

this is my upgraded xdrv_ws2812 files... sorry need to learn github.. :)
```
/*
  xdrv_ws2812.ino - ws2812 led string support for Sonoff-Tasmota

  Copyright (C) 2017  Heiko Krupp and Theo Arends

  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifdef USE_WS2812
/*********************************************************************************************\
 * WS2812 Leds using NeopixelBus library
\*********************************************************************************************/

#include <NeoPixelBus.h>

#ifdef USE_WS2812_DMA
#if (USE_WS2812_CTYPE == 1)
  NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod> *strip = NULL;
#elif (USE_WS2812_CTYPE == 2)
  NeoPixelBus<NeoRgbwFeature, Neo800KbpsMethod>*strip = NULL;
#elif (USE_WS2812_CTYPE == 3)
  NeoPixelBus<NeoGrbwFeature, Neo800KbpsMethod>*strip = NULL;
#else  // USE_WS2812_CTYPE
  NeoPixelBus<NeoRgbFeature, Neo800KbpsMethod> *strip = NULL;
#endif  // USE_WS2812_CTYPE
#else  // USE_WS2812_DMA
#if (USE_WS2812_CTYPE == 1)
  NeoPixelBus<NeoGrbFeature, NeoEsp8266BitBang800KbpsMethod> *strip = NULL;
#elif (USE_WS2812_CTYPE == 2)
  NeoPixelBus<NeoRgbwFeature, NeoEsp8266BitBang800KbpsMethod> *strip = NULL;
#elif (USE_WS2812_CTYPE == 3)
  NeoPixelBus<NeoGrbwFeature, NeoEsp8266BitBang800KbpsMethod> *strip = NULL;
#else  // USE_WS2812_CTYPE
  NeoPixelBus<NeoRgbFeature, NeoEsp8266BitBang800KbpsMethod> *strip = NULL;
#endif  // USE_WS2812_CTYPE
#endif  // USE_WS2812_DMA

struct WsColor {
  uint8_t red, green, blue;
};
struct ColorScheme {
  WsColor* colors;
  uint8_t count;
};
[... 324 lines cut ...]
  return scolor;
}

void Ws2812ShowScheme(uint8_t scheme)
{
  switch (scheme) {
    case 0:  // Clock
      if (((STATES/10)*2 == state) || (ws_show_next)) {
        Ws2812Clock();
        ws_show_next = 0;
      }
      break;
    default:
      if (1 == Settings.light_fade) {
        Ws2812Gradient(scheme -1);
      } else {
        Ws2812Bars(scheme -1);
      }
      ws_show_next = 1;
      break;
  }
}

#endif  // USE_WS2812
```

jhlweb, 2017-11-11:
@arendst  i got the RGBW neopixel ledstrips working with all schemes
changes in xdrv_snfled: 
line 217
```
  light_subtype = light_type &7;
#ifdef (USE_WS2812_CTYPE >1)
  light_subtype = light_subtype + 1; //added jhlweb
#endif
```

line 541
```
if (LT_WS2812 == light_type) {
#if (USE_WS2812_CTYPE > 1)
        Ws2812SetColor(0, cur_col[0], cur_col[1], cur_col[2], cur_col[3]); //added jhlweb
#elif
        Ws2812SetColor(0, cur_col[0], cur_col[1], cur_col[2]);
#endif //USE_WS2812_CTYPE
}
```

line 802
```
if (LightColorEntry(dataBuf, data_len)) {
#if (USE_WS2812_CTYPE > 1)
        Ws2812SetColor(index, light_entry_color[0], light_entry_color[1], light_entry_color[2], light_entry_color[3]); //added jhlweb
#elif 
        Ws2812SetColor(index, light_entry_color[0], light_entry_color[1], light_entry_color[2]);
#endif  //USE_WS2812_CTYPE
}
```

and then there is file: xdrv_ws2812, ill post that completely because there are a lot of minor change
```
/*
  xdrv_ws2812.ino - ws2812 led string support for Sonoff-Tasmota

  Copyright (C) 2017  Heiko Krupp and Theo Arends

  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifdef USE_WS2812
/*********************************************************************************************\
 * WS2812 Leds using NeopixelBus library
\*********************************************************************************************/

#include <NeoPixelBus.h>

#ifdef USE_WS2812_DMA
#if (USE_WS2812_CTYPE == 1)
  NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod> *strip = NULL;
#elif (USE_WS2812_CTYPE == 2)
  NeoPixelBus<NeoRgbwFeature, Neo800KbpsMethod>*strip = NULL;
#elif (USE_WS2812_CTYPE == 3)
  NeoPixelBus<NeoGrbwFeature, Neo800KbpsMethod>*strip = NULL;
#else  // USE_WS2812_CTYPE
  NeoPixelBus<NeoRgbFeature, Neo800KbpsMethod> *strip = NULL;
#endif  // USE_WS2812_CTYPE
[... 353 lines cut ...]
  return scolor;
}

void Ws2812ShowScheme(uint8_t scheme)
{
  switch (scheme) {
    case 0:  // Clock
      if (((STATES/10)*2 == state) || (ws_show_next)) {
        Ws2812Clock();
        ws_show_next = 0;
      }
      break;
    default:
      if (1 == Settings.light_fade) {
        Ws2812Gradient(scheme -1);
      } else {
        Ws2812Bars(scheme -1);
      }
      ws_show_next = 1;
      break;
  }
}

#endif  // USE_WS2812
```

arendst, 2017-11-11:
Thnx. Will look into it. Just ordered my WS2812 RGBW ledstring for verification.

jhlweb, 2017-11-11:
one small error `#ifdef `must be `#if`.. 
line 217
```
  light_subtype = light_type &7;
#if (USE_WS2812_CTYPE >1)
  light_subtype = light_subtype + 1; //added jhlweb
#endif
```

closed 2018-03-11 by jhlweb