arendst/Tasmota #1157: Error: Upload failed, File upload buffer miscompare

derm23, 2017-11-08:
Trying to upgrade a whole bunch of remote devices via their webpages from 5.0.1 to 5.9.1.a and get this error. Searched for the issue and it suggests a two step process first with; 
#define USE_MINIMAL enabled
then #define USE_MINIMAL disabled.
Can't figure out where in the code this line goes. Please help.

derm23, 2017-11-08:
Got it to work by using the minimal firmware but still don't see where #define USE_MINIMAL should be placed.

closed 2017-11-08 by derm23

davidelang, 2017-11-08:
US_MINIMAL is in user_config.h normally.

David Lang

On Wed, 8 Nov 2017, derm23 wrote:

> Date: Wed, 08 Nov 2017 22:21:25 +0000 (UTC)
> From: derm23 <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24fec2b1d079154e428e738166c0cc9b07904286f3892cf00000001161b456
>     292a169ce103bff83@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] Error: Upload failed,
>     File upload buffer miscompare (#1157)
> 
> Got it to work by using the minimal firmware but still don't see where #define USE_MINIMAL should be placed.
>
>

shayshtern, 2017-11-09:
I also go fail when try to upload the sonoff.bin file. ...